blueman-project/blueman #1240: Multiple device fade in not working properly

Yannik, 2020-04-28:
Only one fade-in actually works if there are multiple ones at the same time.

This is similar to the issues for fade-out solved in #1234.

kuraga, 2022-05-02:
Yes.

Have three devices, `Third` is connected, `First` and `Second` aren't. On re-opening Bluetooth Devices dialog, see `First` and `Third`, both "are disconnected".

blueman-applet 2.2.2 on Gentoo.

kuraga, 2022-05-02:
Hmm... Strange catch.

I have an exception [here](https://github.com/blueman-project/blueman/blob/2.2.3/blueman/gui/manager/ManagerDeviceList.py#L408), see #1708. But if I catch (by `except`) this exception situation descripted above dissapears!!!

Race conditions?

Or exception stops adding a device? But connected device exists in the device list (but as "disconnected")!

kuraga, 2022-05-02:
/cc @cschramm

kuraga, 2022-05-02:
And even more:

on turning on `MAJOR IV` headset (MAC `...:6E`) I see:
```
blueman-manager 22.39.15 INFO     ManagerDeviceList:369 row_update_event: Connected True
blueman-manager 22.39.15 DEBUG    ManagerDeviceMenu:297 generate  : JBL E45BT
blueman-manager 22.39.15 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_..._6E {'Connected': True}
```
(BTW, auto-connection is turned off for `MAJOR IV` if I look for menu.)

JBL E45BT!!!!!!

/cc @cschramm