blueman-project/blueman #1241: do_g_properties_changed UnicodeDecodeError on bluetoothd stop

Yannik, 2020-04-28:
When stopping bluetoothd, there is an UnicodeDecodeError without traceback.
This is similar to issue #620. 
It can be easily reproduced using this [script](https://github.com/blueman-project/blueman/issues/620#issuecomment-266291535) from @infirit.

Unfortunately, this even occurs with the `GET_INVALIDATED_PROPERTIES` flag is used, unlike the exception discussed in #620.

infirit, 2020-05-01:
These are the invalidated properties and blueman does not care about those (and probably never will).

Yannik, 2020-05-01:
I know, this issue is about the `UnicodeDecodeError` exceptions.

infirit, 2020-05-01:
We are probably misunderstanding each other so let me elaborate a bit. These UnicodeDecodeErrors are always for invalidated properties on a dbus proxy or objectmanager. The invalidated properties in c is an array which should be translated to a python list. Instead they are introspected as a utf8 bytes and when decoding thing obviously blow up. Blueman does not use the invalidated properties and as you noticed they don't result in an exception. You can safely ignore these errors.

The funny thing is that it **only** happens when subclassing and using the virtual function do_g_properties_changed. Non subclass usage is unaffected, which is probably why very few people care.

Yannik, 2020-05-01:
I did read the upstream glib bug so I'm aware of the introspection issue. You [wrote](https://github.com/blueman-project/blueman/issues/620#issuecomment-266856399) that using the `GET_INVALIDATED_PROPERTIES` flags get's rid of the errors.

According to docs, `GET_INVALIDATED_PROPERTIES` should do the following:

> If set, the property value for any invalidated property will be (asynchronously) retrieved upon receiving the PropertiesChanged D-Bus signal and the property will not cause emission of the g_properties_changed signal. When the value is received the g_properties_changed signal is emitted for the property along with the retrieved value. 

As this is now happening again, I'd assume this problem also occurs with non-invalidated properties..?

Anyway, I simply wanted to give a headsup about this, as contrary to your [comment](https://github.com/blueman-project/blueman/issues/620#issuecomment-266856399), this does not seem fixed in all situations for me.

closed 2020-07-09 by stale[bot]

cschramm, 2020-07-09:
For the records: This is a known glib issue tracked at https://gitlab.gnome.org/GNOME/glib/-/issues/1087 causing a harmless exception in our callback. Probably nothing we can do about it, apart from improving glib itself.