blueman-project/blueman #1242: Failed to change profile to headset_head_unit Ubuntu 18.04.

BionicBeaver74, 2020-05-06:
blueman-applet: 2.0.5
BlueZ: 5.0?
Distribution: Ubuntu 18.04.
Desktop environment:
This error occurs everytime I try to switch my connected and already paired jbl650btnc headphones from A2DP to HSP/HFP (handsfree mode). I know this bug has been around since the big bang but is there ANY SOLUTION AVAILABLE in the foreseeable future? I live in hope...

hugokerstens, 2020-05-06:
I'm having the same problem, this is the same issue as #1031.

As a work-around, you can manually disconnect the headphones and then reconnect them as 'Audio sink', which has audio profile 'A2DP sink' by default.

BionicBeaver74, 2020-05-06:
A2DP is working fine - using my headset for Skype e.g. is the problem...

infirit, 2020-05-06:
Bluetooth headset is a bit of a mess under gnu/linux and unfortunately there is not much we can do from blueman. The only thing blueman does is provide the same switch that common audio mixers have. We just tell pulseaudio to change the profile and check if it was successful, nothing more. If you want help for you distribution i suggest you ask in the Ubuntu forums, there are many more people that can help.

hugokerstens, 2020-05-06:
@BionicBeaver74 Sorry I misread you issue. Still, I think it comes down to same problem.

I did some research and it appears that when the headphone is connected automatically (thus on powering on the headphone) it fails to discover all possible modes, but only sees the default one (which is HSP/HFP in your case).

What I found is that when I connect my headset manually I'm able to switch modes. For specific headsets, using the volume/play/pause controls could also trigger your computer into discovering the other modes. [Source](https://wiki.archlinux.org/index.php/Bluetooth_headset#Switch_between_HSP/HFP_and_A2DP_setting)

Disconnecting and connecting the headset every time it connects can be quite a hassle, but [here is an article describing how to do that automatically](https://medium.com/@aiguofer/flawless-bluetooth-headset-mdr-100abn-on-linux-e745cb746671). This approach is what I've been using now, and I have to say it works quite well.

cschramm, 2020-05-07:
@hugokerstens: Interesting. I guess "connected automatically (thus on powering on the headphone)" means that the headphone device initializes the connection. I'm not sure how you avoid that but if the point is that the PC needs to initialize the connection, the current blueman master has some backend parts for an auto-connect feature that could be used to do that, see https://github.com/blueman-project/blueman/pull/853.

hugokerstens, 2020-05-07:
@cschramm Yeah that's what I meant, the headset initiating the connection.

Apparently it has been fixed the other way around (discovering audio sink when HFP/HSP connects) in [this commit in bluez](https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=477ecca127c529611adbc53f08039cefaf86305d). This commit is present in the 5.51 release of bluez, but currently my distro (debian 10.3) ships with bluez 5.50, so I can't say for sure if the fix works. 

Looking in the commit history, there's no sign of any other fixes for discovering HFP/HSP...

cschramm, 2020-05-08:
At least in https://github.com/blueman-project/blueman/issues/1209#issuecomment-625155850 that improvement actually showed the desired results.

So, @BionicBeaver74, you need to update your version of BlueZ. Easiest solution is probably to switch to Ubuntu 20.04.

closed 2020-05-08 by cschramm

BionicBeaver74, 2020-05-08:
Thanks! Is there a catch just updating bluez? Because my Laptop which is an Acer Extensa 5220 with 2GB of RAM upgraded to the max of specs memorywise and with a new SSD could be too slow for 20.04. It’s running fine now. What would be the right procedure to just update bluez and stick with 18.04.? Would there be dependency issues?

cschramm, 2020-05-08:
@BionicBeaver74: You could just download [the 20.04 package](https://packages.ubuntu.com/focal/bluez) and install it with `dpkg -i`. To satisfy dependencies you'll also need `libreadline8` and `libtinfo6`. Everything else should fit. Just keep in mind that you will not receive any updates if you do it manually like that.

BionicBeaver74, 2020-05-08:
Just installed bluez 5.50 under 18.04. but still the same issue: my headphones autoconnect to my bluetooth dongle (which is fine) as A2DP. When I try to switch to handsfree blueman says: Failed to change profile to headset_head_unit. I give up - but thx anyway for your time and quick response, you guys are great!

cschramm, 2020-05-09:
You'd need at least 5.31. 20.04 has 5.53.

lucone83, 2021-03-28:
Same problem here with bluez 5.53 and Ubuntu 20.04. My headphones autoconnects as A2DP. When I try to switch to handsfree blueman says: Failed to change profile to headset_head_unit.

andrii-pukhalevych, 2021-04-13:
@cschramm Also bluez 5.53 and Ubuntu 20.04 returns error `Failed to change profile to headset_head_unit`

[image]