blueman-project/blueman #1243: New release ?

joakim-tjernlund, 2020-05-07:
Lots of commits since last release, could we have a new release please?

cschramm, 2020-05-07:
Are you talking about [2.2](https://github.com/blueman-project/blueman/milestones/2.2) or a [2.1.3](https://github.com/blueman-project/blueman/blob/5af71628bbdb34a6a1614a5621c3c8741c9f130c/CHANGELOG.md#213)? 2.2 is far from ready, as you can see, a 2.1.3 is possible. Which changes do you need released?

joakim-tjernlund, 2020-05-07:
My blueman opens the Devices window in the background, behind all other windows.
I was hoping that was fixed in git

infirit, 2020-05-07:
There is no fix for this in master. I did notice this and it may be because
of the wm focus stealing logic. Open a new issue for this so we can
properly track it.

As for a new release, there are a few fixes in the stable branch which
could be released into the wild :-)

cschramm, 2020-05-08:
There we go: https://github.com/blueman-project/blueman/releases/tag/2.1.3 :slightly_smiling_face:

closed 2020-05-08 by cschramm