blueman-project/blueman #1247: Bluettoth software recognizes my hardware on all Linux systems except latest LXLE

Ahhh-Clem, 2020-05-08:
blueman:
BlueZ:
Distribution:
Desktop environment:

I have been using different versions of Linux as a trial on several machines as a trial.  When I installed the latest versions of LXLE on different machines,  I encountered a problem in common.  On each of these machines after booting,  the software could not connect to my bluetooth device until I removed it and reinserted the device. Immediately everything was fine.  
This has to be done each time I start the PC again.  This does not happen on any of my other operating systems which are using this software or the same bluetooth hardware.
Needless to say, this is rather annoying.
I have tried completely uninstalling the software.  I tracked down the configuration file and deleted it.
I have even started a completely new installation of the entire operating system.
The only other questionable detail I have noted of late is the LXLE problem with Linux 4.15.0 in which each time it is upgraded the user recieves a warning about removing the kernal which is 'already in use'.
Linux 4.15.0-99 generic
Ubuntu 18.04.4 LTS

cschramm, 2020-05-09:
Please visit the [Troubleshooting](https://github.com/blueman-project/blueman/wiki/Troubleshooting) page you ignored and provide a log for the failing connection attempt.

What do you remove and re-insert? If that refers to a USB bluetooth controller, I guess there's some udev rules that get you ready when you plug it in. It should not be about powering the adapter, though, as Debian and Ubuntu both use BlueZ' `AutoEnable=true`.

Ahhh-Clem, 2020-07-10:
Thank you.  Did not ignore I am working to death here with no time for
anything else.  I think that I looked at what was there but perhaps was too
frazzled to be productive about it.  You are generous with your help.  I
tried but could not find the log.

Have to try to catch a few zzzzz's I have been up with power outages on
major equipment in some HUD buildings.
Thank you again.
Dan

closed 2020-10-09 by github-actions[bot]