blueman-project/blueman #1248: blueman-services does not allow a user to save transfer settings

S4qFBxkFFg, 2020-05-10:
blueman: 2.1.5-1 (focal)
BlueZ: 5.53-3ubuntu3 (focal)
Distribution: Ubuntu 20.04 (focal)
Desktop environment: XFCE 4.14

When the blueman applet is enabled, right-clicking and selecting "Local Services..." then "Transfer" does not allow a new destination folder to be selected and saved.

Instead, the new folder can be selected, but not saved as the "Apply" button is greyed out.

When running the command as root, the "Apply" button is available, but this only affects the root user.

I just tried upgrading to the latest version of Ubuntu, but the problem remains.  I also tried creating a new user on this machine, and the same problem resulted.

terminal log from starting the applet, attempting to change destination folder, and exiting:
https://pastebin.com/7kz9fhdy

Please let me know if any logs are required; also reported at: https://bugs.launchpad.net/ubuntu/+source/blueman/+bug/1877131

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2020-05-13:
> blueman: 2.1.5-1 (focal)
Should be 2.1.2 I guess.

As a first try, please apply this patch:

```diff
--- /usr/lib/python3/dist-packages/blueman/plugins/services/Transfer.py
+++ /usr/lib/python3/dist-packages/blueman/plugins/services/Transfer.py
@@ -42,6 +42,7 @@
         self.widget.props.visible = False
 
     def on_property_changed(self, config, key):
+        logging.debug(key)
         value = config[key]
 
         if key == "shared-path":
```

run `blueman-services --loglevel debug`, replicate the issue, and provide the output.

S4qFBxkFFg, 2020-05-14:
> Should be 2.1.2 I guess.

Apologies - typo when updating the post after upgrading Ubuntu - corrected.

The requested output is:
```
blueman-services --loglevel debug
blueman-services version 2.1.2 starting
blueman-services 11.43.04 INFO     Services:90 load_plugins: ['Network', 'Transfer']
blueman-services 11.43.05 INFO     Services:128 set_page  : Set page Network
blueman-services 11.43.05 INFO     Network:38 on_load   : enp4s0
blueman-services 11.43.09 INFO     Services:128 set_page  : Set page Transfer
```
No logs are generated in the terminal when I change the value in the "Incoming Folder:" field, and the "Apply" button stays greyed/inactive.

infirit, 2020-05-14:
The apply button here is pretty useless as the value is already set when you select the path. What does ``gsettings get org.blueman.transfer shared-path`` from a terminal show?

S4qFBxkFFg, 2020-05-14:
It looks like an empty string:
```
 gsettings get org.blueman.transfer shared-path
''
```
When I change the path in blueman-services, it doesn't affect the output of the gsettings command if I do it again.

S4qFBxkFFg, 2020-05-14:
Hmm - if I try to use `gsettings set` to change that value, it says `The key is not writable` - permissions issue...?

infirit, 2020-05-14:
That would indeed be problematic. Can you check your permission on the dconf file (this is were settings are stored) with ``ls -l ~/.config/dconf/user``. If the permission is wrong update it how they should be for ubuntu.

I'll open a PR that actually handles the apply button properly. This won't effect your problem but will make the ui consistent.

S4qFBxkFFg, 2020-05-14:
That file doesn't exist - I created an empty file with that path and did chmod 644 on it, but I don't think it's changed anything.

S4qFBxkFFg, 2020-05-15:
So I'm pretty sure this is a problem with dconf - not blueman - because other (unrelated) settings can't be changed either.  I'm assuming I should close this?
Reported here instead: https://bugs.launchpad.net/ubuntu/+source/dconf-editor/+bug/1878966

S4qFBxkFFg, 2020-05-15:
Solved!
For some reason, dconf was attempting to save settings in `/var/lib/snapd/desktop/dconf/profile/user` which was only writable by root, removing this file made dconf work correctly.
I discovered this by doing an strace on the dconf command:
`strace -o trace dconf write /org/blueman/transfer/shared-path "'/home/me/download/bluetooth'"`
then grepping the output for files opened:
`grep openat trace`
This line was the clue:
`openat(AT_FDCWD, "/var/lib/snapd/desktop/dconf/profile/user", O_RDONLY) = -1 ENOENT (No such file or directory)`

Sorry for implicating blueman here.

closed 2020-05-15 by S4qFBxkFFg

infirit, 2020-05-16:
> Sorry for implicating blueman here.

No worries, I actually added a fix and a log when dconf is not writable based on our discussion here.

>/var/lib/snapd/desktop/dconf/profile/user

This look like a proper ubuntu bug to me that needs to be investigated. Normally you as a users should never look for your dconf file in /var. ``snapd`` suggest the ubuntu package manager was involved somehow.