blueman-project/blueman #1251: Logitech MX Master mouse fails to pair

mareg, 2020-05-12:
blueman:
```
blueman-2.1.3-1.fc31.x86_64
```
BlueZ:
```
bluez-5.54-1.fc31.x86_64
bluez-libs-5.54-1.fc31.x86_64
bluez-tools-0.2.0-0.11.git20170912.7cb788c.fc31.x86_64
bluez-obexd-5.54-1.fc31.x86_64
```
Distribution: Fedora 31
Desktop environment: MATE desktop

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

No matter what I do, it does not pair my device. On another computer (laptop) I have no problem pairing this, or another Logitech mouse. But on my desktop I get the authentication error.

[image]

When I try to use the `bluetoothctl` console, I get the following:
```
$ bluetoothctl
Agent registered
[bluetooth]# agent on
Agent is already registered
[bluetooth]# devices
Device DD:51:7A:19:7D:AD MX Master
[CHG] Device DD:51:7A:19:7D:AD RSSI: -94
[bluetooth]# pair DD:51:7A:19:7D:AD
Attempting to pair with DD:51:7A:19:7D:AD
[CHG] Device DD:51:7A:19:7D:AD Connected: yes
Failed to pair: org.bluez.Error.AuthenticationFailed
[CHG] Device DD:51:7A:19:7D:AD Connected: no
```
relevant log from the command line:
```
blueman-assistant 16.09.58 WARNING  DeviceList:193 monitor_power_levels: Failed to get power levels, probably a LE device.
blueman-applet 16.09.58 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_DD_51_7A_19_7D_AD {'Connected': True}
blueman-manager 16.09.58 WARNING  DeviceList:193 monitor_power_levels: Failed to get power levels, probably a LE device.
blueman-assistant 16.09.58 ERROR    blueman-assistant:197 err       :  Authentication Failed
blueman-applet 16.09.58 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_DD_51_7A_19_7D_AD {'Connected': False}
blueman-applet 16.10.43 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Discovering': False}
blueman-applet 16.10.43 DEBUG    DiscvManager:82 on_adapter_property_changed: prop Discovering False
```

cschramm, 2020-05-13:
You could try `default-agent` in `bluetoothctl` to replace the default agent and see if something comes up with `pair DD:51:7A:19:7D:AD` then. `bluetoothd` logs might also give some hint.

mareg, 2020-05-14:
Same result with `default-agent` in `bluetoothctl`:
```
$ bluetoothctl
Agent registered
[bluetooth]# default-agent
Default agent request successful
[bluetooth]# devices
Device 28:C1:3C:CD:C3:07 Wireless Controller
Device DD:51:7A:19:7D:AD MX Master
[bluetooth]# pair DD:51:7A:19:7D:AD
Attempting to pair with DD:51:7A:19:7D:AD
[CHG] Device DD:51:7A:19:7D:AD Connected: yes
Failed to pair: org.bluez.Error.AuthenticationFailed
[CHG] Device DD:51:7A:19:7D:AD Connected: no
[bluetooth]# pair 28:C1:3C:CD:3C:07
Device 28:C1:3C:CD:3C:07 not available
[bluetooth]# pair 28:C1:3C:CD:3C:07
Device 28:C1:3C:CD:3C:07 not available
```
There seem to be not much in `bluetoothd` logs, it only mentions my Wireless controller being connected, nothing re MX Master:
```
May 14 16:22:07 stefan bluetoothd[1153]: sixaxis: compatible device connected: Wireless Controller (054C:09CC /sys/devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.3/0003:054C:09CC.000D/hidraw/hidraw0)
May 14 16:22:07 stefan bluetoothd[1153]: sixaxis: setting up new device
May 14 16:22:07 stefan bluetoothd[1153]: sixaxis: compatible device connected: Wireless Controller (054C:09CC /sys/devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.3/0003:054C:09CC.000D/hidraw/hidraw0)
May 14 16:22:07 stefan bluetoothd[1153]: sixaxis: setting up new device
```

infirit, 2020-05-14:
What happens if you mark the device as trusted and connect, skipping pairing?

cschramm, 2020-05-15:
You could stop the bluetooth service and start `bluetoothd` with the `-d` flag to get debugging output.

mareg, 2020-05-25:
> What happens if you mark the device as trusted and connect, skipping pairing?

That was one of the first things I've tried. Same error.

> You could stop the bluetooth service and start `bluetoothd` with the `-d` flag to get debugging output.
I've tried to stop and start it, seems like I don't have this service.
```
[root@localhost ~]# service bluetoothd stop
Redirecting to /bin/systemctl stop bluetoothd.service
Failed to stop bluetoothd.service: Unit bluetoothd.service not loaded.
[root@localhost ~]# bluetoothd -d
-bash: bluetoothd: command not found
[root@localhost ~]# service bluetoothd start
Redirecting to /bin/systemctl start bluetoothd.service
Failed to start bluetoothd.service: Unit bluetoothd.service not found.
[root@localhost ~]# dnf install bluetoothd
Last metadata expiration check: 0:21:16 ago on Mon 25 May 2020 20:08:47 BST.
No match for argument: bluetoothd
Error: Unable to find a match: bluetoothd
[root@localhost ~]# dnf install bluetooth
Last metadata expiration check: 0:21:22 ago on Mon 25 May 2020 20:08:47 BST.
No match for argument: bluetooth
Error: Unable to find a match: bluetooth
[root@localhost ~]# 
```
I've changed my drive and had to install a fresh Fedora 32. I kind of hoped the issue would go away. Unfortunately, it didn't.

buildog, 2020-06-02:
I do have the same issue, which in my case could be caused by security policies

**bluetoothctl**
```
pi@raspberrypi:/etc/dbus-1/system.d $ sudo bluetoothctl
Agent registered
[bluetooth]# devices
Device 9C:20:7B:BE:14:6E 9C-20-7B-BE-14-6E
Device DA:39:F5:DD:38:C8 N030M
[bluetooth]# scan on
Discovery started
[CHG] Controller B8:27:EB:08:5F:F0 Discovering: yes
[CHG] Device 9C:20:7B:BE:14:6E RSSI: -70
[CHG] Device 9C:20:7B:BE:14:6E ManufacturerData Key: 0x004c
[CHG] Device 9C:20:7B:BE:14:6E ManufacturerData Value:
  09 06 02 0f c0 a8 01 03                          ........        
[NEW] Device 46:C2:C2:1C:0B:A9 46-C2-C2-1C-0B-A9
[CHG] Device DA:39:F5:DD:38:C8 RSSI: -71
[CHG] Device DA:39:F5:DD:38:C8 UUIDs: 0000feaf-0000-1000-8000-00805f9b34fb
[CHG] Device DA:39:F5:DD:38:C8 ServiceData Key: 0000feaf-0000-1000-8000-00805f9b34fb
[CHG] Device DA:39:F5:DD:38:C8 ServiceData Value:
  10 01 00 02 5a 23 09 00 e5 bc 9d 00 00 30 b4 18  ....Z#.......0..
  00                                               .               
[NEW] Device C4:64:E3:9B:04:44 315X
[NEW] Device E9:E6:B6:6D:DA:49 N065R
[bluetooth]# scan off
Discovery stopped
[CHG] Controller B8:27:EB:08:5F:F0 Discovering: no
[CHG] Device E9:E6:B6:6D:DA:49 RSSI is nil
[CHG] Device C4:64:E3:9B:04:44 RSSI is nil
[CHG] Device DA:39:F5:DD:38:C8 RSSI is nil
[CHG] Device 46:C2:C2:1C:0B:A9 TxPower is nil
[CHG] Device 46:C2:C2:1C:0B:A9 RSSI is nil
[CHG] Device 9C:20:7B:BE:14:6E RSSI is nil
[bluetooth]# agent on
Agent is already registered
[bluetooth]# default-agent
Default agent request successful
[bluetooth]# pair C4:64:E3:9B:04:44
Attempting to pair with C4:64:E3:9B:04:44
[CHG] Device C4:64:E3:9B:04:44 Connected: yes
Failed to pair: org.bluez.Error.AuthenticationFailed
[CHG] Device C4:64:E3:9B:04:44 Connected: no
```

**bluetoothd** 
```
pi@raspberrypi:~/bergepi $ bluetoothd
D-Bus setup failed: Connection ":1.41" is not allowed to own the service "org.bluez" due to security policies in the configuration file
``` 

***/etc/dbus-1/system.d/bluetooth.conf***
```
<!-- This configuration file specifies the required security policies
     for Bluetooth core daemon to work. -->

<!DOCTYPE busconfig PUBLIC "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>

  <!-- ../system.conf have denied everything, so we just punch some holes -->

  <policy user="root">
    <allow own="org.bluez"/>
    <allow send_destination="org.bluez"/>
    <allow send_interface="org.bluez.Agent1"/>
    <allow send_interface="org.bluez.MediaEndpoint1"/>
    <allow send_interface="org.bluez.MediaPlayer1"/>
    <allow send_interface="org.bluez.Profile1"/>
    <allow send_interface="org.bluez.GattCharacteristic1"/>
    <allow send_interface="org.bluez.GattDescriptor1"/>
    <allow send_interface="org.bluez.LEAdvertisement1"/>
    <allow send_interface="org.freedesktop.DBus.ObjectManager"/>
    <allow send_interface="org.freedesktop.DBus.Properties"/>
  </policy>

  <!-- allow users of bluetooth group to communicate -->
  <policy group="bluetooth">
    <allow send_destination="org.bluez"/>
  </policy>

  <policy at_console="true">
    <allow send_destination="org.bluez"/>
  </policy>

  <!-- allow users of lp group (printing subsystem) to
       communicate with bluetoothd -->
  <policy group="lp">
    <allow send_destination="org.bluez"/>
  </policy>

  <policy context="default">
    <deny send_destination="org.bluez"/>
  </policy>

</busconfig>
```

mareg, 2020-06-02:
In my case, I have 2 devices - desktop and laptop - both running Fedora 32 with MATE desktop. I have no issues getting said mouse or bose headphones connected to my laptop, but no luck with my desktop. I'd assume both devices have same default policies, but I'll check that.

cschramm, 2020-06-15:
> > You could stop the bluetooth service and start `bluetoothd` with the `-d` flag to get debugging output.
> 
> I've tried to stop and start it, seems like I don't have this service.

The service is called `bluetooth` and the binary is usually at `/usr/sbin/bluetoothd` and probably not in your path and you will need to run it as root.

ptahchiev, 2021-01-29:
I have the exact same issue on Dell XPS 15 9550, Ubuntu 20.04.1:
```
Linux petar-XPS-15-9550 5.4.0-64-generic #72-Ubuntu SMP Fri Jan 15 10:27:54 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```
```
[bluetooth]# pair 48:2C:A0:1F:F1:0A
Attempting to pair with 48:2C:A0:1F:F1:0A
Failed to pair: org.bluez.Error.AuthenticationCanceled
```
Has any of you found a solution?

DrCWO, 2021-04-07:
Same Issue on Pi4
Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux

[bluetooth]# pair 70:BC:10:85:CB:01
Attempting to pair with 70:BC:10:85:CB:01
[CHG] Device 70:BC:10:85:CB:01 Connected: yes
Request authorization
[agent] Accept pairing (yes/no): y
Failed to pair: org.bluez.Error.AuthenticationFailed
[CHG] Device 70:BC:10:85:CB:01 Connected: no

sartyukhov, 2021-07-07:
Hi! Same on my Xiaomi RedmiBook 16 (Ryzen 4600) with Manjaro KDE.
It's so strange when perfectly working systems can`t do basic things.

ltchurch88t, 2021-10-15:
For what it's worth I got my MX Master 3 to pair with my Pi3. Ended up pulling in the pulse audio package then rebooting system and connecting via bluetoothctl. Steps below:

`sudo apt install pulseaudio-module-bluetooth`

reboot system

`
pi@raspberrypi:~ $ bluetoothctl
Agent registered
[bluetooth]# power on
[bluetooth]# agent on
[bluetooth]# default-agent
[bluetooth]# scan on
[bluetooth]# scan off
[bluetooth]# trust DA:8B:95:0D:92
[bluetooth]# pair DA:8B:95:0D:92
`

After these steps device connected and paired successfully. Not sure if it will help anyone else or not!

ismaell, 2022-04-11:
This seems to be a common problem with pointing devices, I've had the same issue with the Logitech G603 and the MX, and a suspiciously similar one with Wacom tablets. They pair fine on Android 9 to 11, so it's probably just the agent. Tried with several controllers on several machines, same issue.

ismaell, 2022-04-11:
I found out in my case the problem was caused by a quirky bluetooth controller.

The controller was an UGREEN CM390 (Realtek 8761BUV, USB).

Fix here: https://patchwork.kernel.org/project/bluetooth/patch/20220411174747.13704-1-ismael@iodev.co.uk/