blueman-project/blueman #1255: Mouse Bluetooth freeze, cannot remove or reinstall Xubuntu 20.04

Phil171, 2020-05-15:
Xubuntu 20.04
Blueman: blueman-manager 2.1.2
Xfce
mouse Bluetooth Logitech M535 freeze or stops working, cannot remove it in blueman-manager or de-install, reinstall it. New battery.
Logitech M525 on USB dongle works perfectly.
Obligation to restart the computer.
Philippe from France,
Sorry for bad english or missing information

xrombik, 2020-06-25:
I have same issue with another BT mouse device on another Linux (e.g. MX Linux with 4.19.128 kernel). My workaround is just restart **btusb** driver and **bluetooth** service (enter in terminal):
```
$ sudo service bluetooth stop
$ sudo modprobe -r btusb
$ sudo modprobe btusb
$ sudo service bluetooth start
```
Then mouse work perfect about 3-4 hours, and then i repeat this commands if something goes wrong. I think this is a **btusb** module related problem.

cschramm, 2020-06-26:
Xubuntu 20.04 has a completely different kernel version but it sounds like a driver issue, yes. @Phil171, does restarting bluetoothd and / or reloading the driver module fix it?

Phil171, 2020-06-26:
> Xubuntu 20.04 has a completely different kernel version but it sounds like a driver issue, yes. @Phil171, does restarting bluetoothd and / or reloading the driver module fix it?

Hi, cschramm,
I was using those commands in Terminal to restart Bluetooth :
`sudo service bluetooth restart && sudo modprobe -r btusb && sudo modprobe btusb`
It was restarting Wi-fi and BT card (it is a Killer 1650, Wi-Fi and BT device).

But yet, after using last kernel drivers it seems to be OK, sometime a little lag when starting Xubuntu, but after, no problem.
My kernel is **5.4.0-37-generic** On Ubuntu 20.04 LTS with Xfce.
Hope this help, thanks !

cschramm, 2020-07-04:
If I get you right the problem is gone with a newer kernel, so there's not even a need to restart or reload anything and you problem is solved, right?

Phil171, 2020-07-05:
Yes, You are right. NO need to restart computer or reload something, program or service.
Just a little freeze after Xubuntu starts, I mean Xubuntu is coming, Desktop appears, BT is established, then mouse works and can move, then stops (or freeze) for a few seconds and after this small "stop" it is OK, it works without problem.
Now I'm working with a new mouse, I have replaced my **M535** to a **Logitech MX Anywhere 2** because multi channel (2x BT, and one with "Logitech Unifying" dongle), but it seems that M535 was working perfectly, I've replaced under warranty because of a materiel problem (soft plastic or silicone constituting the external body of the mouse had become very sticky).

I've found another problem, but I cannot determinate if it is from mouse or OS or Thunar explorer, I'll investigate later.

FallenPX, 2020-07-06:
> Just a little freeze after Xubuntu starts, I mean Xubuntu is coming, Desktop appears, BT is established, then mouse works and can move, then stops (or freeze) for a few seconds and after this small "stop" it is OK, it works without problem.

I have the same problem with my M590. The Blueman-applet kind of restarting itself moments after login.

xrombik, 2020-07-07:
> Yes, You are right. NO need to restart computer or reload something, program or service.
> Just a little freeze after Xubuntu starts, I mean Xubuntu is coming, Desktop appears, BT is established, then [...]

How long do you work continuously at your computer using your BT mouse? We need to understand: the problem occurs randomly or predictably after a certain period of continuous operation.

Phil171, 2020-07-08:
Hi, xrombik, I work from 9:00 AM to 13:00, then from 03:00 PM (15:00) to 08:30 (20:30PM).
The problem occurs EVERY time I start the computer. After, I mean, around 5 minute after start, no problem, no freeze at all.
I specify that, having gone to version 20.04 for reasons of Bluetooth and WiFi drivers included in the kernel of this version for my Dell XPS 15-7590, freeze or blocking faults in Bluetooth were frequent and random during use from the computer in the beginning of this version (and was not appearing so frequently in V18.04, but in this V18.04, I must had have downloaded drivers from Killer or Intel website on an USB stick before to install them apart).
After the various updates, and (apparently kernel updated automatically) of the kernel, these defects became more and more absent and only now this 'freezing' that I explain above occurs.
There is just another problem in Thunar when I "right-click" on a file or directory, but it is difficult for me to explain it in words, I will try to make a video of this problem.
By not knowing if this last problem is related to the BT mouse or a problem concerning version 20.04 or Thunar ...

Phil171, 2020-07-22:
Hello,
One point that I forgot to mention is that when the computer starts up, the BT mouse works and then stops for a few moments then starts working again and there is no BT problem afterwards as described above.
OKAY.
But when there is this 'freeze', (I am under Xubuntu), the Bluetooth icon loses the lock padlock when I no longer can move the mouse, and when I recover the mouse, the padlock reappears. next to the BT icon in the Dashboard bar. The BT icon does not disappear, only the padlock and at the same time loss of mouse use temporarily.
Hope this help.

Thank you for your help ! Philippe from France

Phil171, 2020-09-21:
Hi, 
For me, the problem is not close, I have always  a freeze on my mouse in BT on Xubuntu 20.04 and not in Windows.
If I use a dongle (2,4GHz) as Logitech provide with, no freeze...
What can I do or report as message from Terminal to help to find an issue to this problem? Give me code to enter in Terminal, I'll return messages from it.
Thanks,
Best Regards from France,
Philippe

cschramm, 2020-09-21:
I know that phenomenon for Logitech mice and neither know the reason nor a solution for it, but in any case, it's not a blueman thing. It goes with some "bt_uhid_send: Invalid argument (22)" messages for me and my guess would be some power management quirks that you could e.g. tweak with TLP.

cschramm, 2020-09-21:
Oh, strike that, I was talking about a temporary freeze, but it's probably related closely. There are some references regarding your issue in Ubuntu 20.04 on askubuntu but no related bug report, it seems. You should probably open one.

closed 2020-09-21 by cschramm

Phil171, 2020-09-22:
Thanks Christopher, I'll check on askubuntu.
On TLP, no BT and USB devices are in "eco" or reduced mode.