blueman-project/blueman #1260: Mouseover text / tooltip inaccurate

areographe, 2020-05-19:
```
System:    Kernel: 5.4.31-1-MANJARO x86_64 bits: 64 Desktop: Cinnamon 4.4.8 
           Distro: Manjaro Linux 
Machine:   Type: Laptop System: LENOVO product: 20FC000RAU v: ThinkPad X1 Carbon 4th
           Mobo: LENOVO model: 20FC000RAU UEFI: LENOVO 
           v: N1FET71W (1.45 ) date: 12/10/2019 
CPU:       Topology: Dual Core model: Intel Core i5-6300U bits: 64 type: MT MCP 
           L2 cache: 3072 KiB 
           Speed: 524 MHz min/max: 400/3000 MHz Core speeds (MHz): 1: 549 2: 574 3: 551 4: 528 
Graphics:  Device-1: Intel Skylake GT2 [HD Graphics 520] driver: i915 v: kernel 
           Display: x11 server: X.Org 1.20.8 driver: intel unloaded: modesetting 
           resolution: 1920x1080~60Hz 
           OpenGL: renderer: Mesa Intel HD Graphics 520 (SKL GT2) v: 4.6 Mesa 20.0.4 
Audio:     Device-1: Intel Sunrise Point-LP HD Audio driver: snd_hda_intel 
           Sound Server: ALSA v: k5.4.31-1-MANJARO 
```
When I mouseover the applet icon in the Cinnamon System Tray, the tooltip text is always either:

```
Bluetooth enabled
<b>1 Active Connection</b>
```
or
```
Bluetooth disabled
<b>1 Active Connection</b>
```

The second line is generally inaccurate, as I always have either 0 or 2 devices connected.

Also, the <b> tags are legible without changing the formatting of the text, but this may be a Cinnamon-specific issue.

cschramm, 2020-05-20:
This is a a duplicate of #1163 and fixed in 2.1.3 and in master.

Regarding the markup, there are two different implementations of our status indicator. One based on AppIndicator if the respective plugin is enabled and one based on GtkStatusIcon which is used otherwise. For GtkStatusIcon this would be a bug in Cinnamon's tray implementation as we're even using a dedicated method for setting markup and not only text. For AppIndicator the situation is rather unclear due to a lack of documentation and specification.

areographe, 2020-05-20:
Ah, my fault for not finding #1163 myself, sorry! I am using AppIndicator so I'll look into that. Thank you!

closed 2020-05-20 by areographe