blueman-project/blueman #1262: Sony Vaio. Bluetooth does not find any devices.

supercymek, 2020-05-20:
[code]
System:    Host: **************-SVF1521A6EW Kernel: 5.3.0-53-generic x86_64 bits: 64 compiler: gcc 
           v: 7.5.0 Desktop: Cinnamon 4.4.8 wm: muffin dm: LightDM Distro: Linux Mint 19.3 Tricia 
           base: Ubuntu 18.04 bionic 
Machine:   Type: Laptop System: Sony product: SVF1521A6EW v: C10J0B6L serial: <filter> Chassis: 
           type: 10 serial: <filter> 
           Mobo: Sony model: VAIO serial: <filter> UEFI [Legacy]: Insyde v: R0250DA 
           date: 07/19/2016 
Battery:   ID-1: BAT1 charge: 0 Wh condition: 7.1/7.1 Wh (100%) volts: 3.3/0.7 model: SONY Corp. 
           serial: <filter> status: Charging 
CPU:       Topology: Dual Core model: Intel Pentium 2117U bits: 64 type: MCP arch: Ivy Bridge 
           rev: 9 L2 cache: 2048 KiB 
           flags: lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx bogomips: 7183 
           Speed: 1796 MHz min/max: 800/1800 MHz Core speeds (MHz): 1: 1453 2: 1771 
Graphics:  Device-1: Intel 3rd Gen Core processor Graphics vendor: Sony driver: i915 v: kernel 
           bus ID: 00:02.0 chip ID: 8086:0156 
           Display: x11 server: X.Org 1.20.5 driver: modesetting unloaded: fbdev,vesa 
           resolution: 1366x768~60Hz 
           OpenGL: renderer: Mesa DRI Intel Ivybridge Mobile v: 4.2 Mesa 19.2.8 compat-v: 3.0 
           direct render: Yes 
Audio:     Device-1: Intel 7 Series/C216 Family High Definition Audio vendor: Sony 
           driver: snd_hda_intel v: kernel bus ID: 00:1b.0 chip ID: 8086:1e20 
           Sound Server: ALSA v: k5.3.0-53-generic 
Network:   Device-1: Broadcom and subsidiaries BCM43142 802.11b/g/n vendor: Foxconn driver: wl 
           v: kernel port: 3040 bus ID: 07:00.0 chip ID: 14e4:4365 
           IF: wlp7s0 state: up mac: <filter> 
           Device-2: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet vendor: Sony 
           driver: r8169 v: kernel port: 2000 bus ID: 0e:00.0 chip ID: 10ec:8168 
           IF: enp14s0 state: down mac: <filter> 
           IF-ID-1: pan1 state: unknown speed: N/A duplex: N/A mac: <filter> 
Drives:    Local Storage: total: 465.76 GiB used: 25.65 GiB (5.5%) 
           ID-1: /dev/sda vendor: HGST (Hitachi) model: HTS545050A7E380 size: 465.76 GiB 
           speed: 3.0 Gb/s serial: <filter> 
Partition: ID-1: / size: 29.57 GiB used: 7.76 GiB (26.2%) fs: ext4 dev: /dev/sda10 
           ID-2: /home size: 66.84 GiB used: 17.86 GiB (26.7%) fs: ext4 dev: /dev/sda8 
           ID-3: swap-1 size: 4.11 GiB used: 32.0 MiB (0.8%) fs: swap dev: /dev/sda9 
USB:       Hub: 1-0:1 info: Full speed (or root) Hub ports: 2 rev: 2.0 chip ID: 1d6b:0002 
           Hub: 1-1:2 info: Intel Integrated Rate Matching Hub ports: 4 rev: 2.0 
           chip ID: 8087:0024 
           Device-1: 1-1.2:3 info: Foxconn / Hon Hai type: <vendor specific> driver: btusb 
           rev: 2.0 chip ID: 0489:e062 
           Device-2: 1-1.3:4 info: Realtek type: Video driver: uvcvideo rev: 2.0 
           chip ID: 0bda:5729 
           Hub: 2-0:1 info: Full speed (or root) Hub ports: 2 rev: 2.0 chip ID: 1d6b:0002 
           Hub: 2-1:2 info: Intel Integrated Rate Matching Hub ports: 4 rev: 2.0 
           chip ID: 8087:0024 
           Hub: 3-0:1 info: Full speed (or root) Hub ports: 4 rev: 2.0 chip ID: 1d6b:0002 
           Device-3: 3-2:2 info: Microsoft type: Keyboard,Mouse,HID driver: hid-generic,usbhid 
           rev: 2.0 chip ID: 045e:07b2 
           Hub: 4-0:1 info: Full speed (or root) Hub ports: 4 rev: 3.0 chip ID: 1d6b:0003 
Sensors:   System Temperatures: cpu: 50.0 C mobo: N/A 
           Fan Speeds (RPM): N/A 
Repos:     No active apt repos in: /etc/apt/sources.list 
           Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list 
           1: deb http: //packages.linuxmint.com tricia main upstream import backport #id:linuxmint_main
           2: deb http: //archive.ubuntu.com/ubuntu bionic main restricted universe multiverse
           3: deb http: //archive.ubuntu.com/ubuntu bionic-updates main restricted universe multiverse
           4: deb http: //archive.ubuntu.com/ubuntu bionic-backports main restricted universe multiverse
           5: deb http: //security.ubuntu.com/ubuntu/ bionic-security main restricted universe multiverse
           6: deb http: //archive.canonical.com/ubuntu/ bionic partner
           Active apt repos in: /etc/apt/sources.list.d/steam.list 
           1: deb [arch=amd64,i386] http: //repo.steampowered.com/steam/ precise steam
           2: deb-src [arch=amd64,i386] http: //repo.steampowered.com/steam/ precise steam
Info:      Processes: 195 Uptime: 33m Memory: 3.73 GiB used: 2.20 GiB (59.0%) Init: systemd v: 237 
           runlevel: 5 Compilers: gcc: 7.5.0 alt: 7 Client: Unknown python3.6 client inxi: 3.0.32 
[/code]

->

infirit, 2020-05-20:
Start blueman-applet from the terminal as described in https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-blueman

Then try to search and when done provide the log from the terminal.

supercymek, 2020-05-21:
supercymek@supercymek-SVF1521A6EW:~$ killall blueman-applet
supercymek@supercymek-SVF1521A6EW:~$ blueman-applet loglevel debug
blueman-applet version 2.0.5 starting
Stale PID, overwriting
_________
Load (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:60)
['DiscvManager', 'PPPSupport', 'NetUsage', 'DhcpClient', 'RecentConns', 'AuthAgent', 'StatusIcon', 'KillSwitch', 'Headset', 'Menu', 'ExitItem', 'SerialManager', 'Networking', 'DBusService', 'ShowConnected', 'NMPANSupport', 'GameControllerWakelock', 'AppIndicator', 'TransferService', 'StandardItems', 'PowerManager', 'NMDUNSupport'] 
_________
get_interface_version (/usr/lib/python3/dist-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
/usr/lib/python3/dist-packages/blueman/plugins/applet/AppIndicator.py:8: PyGIWarning: AppIndicator3 was imported without specifying a version first. Use gi.require_version('AppIndicator3', '0.1') before import to ensure that the right version gets loaded.
  from gi.repository import AppIndicator3 as girAppIndicator
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
_________
update_menuitems (/usr/lib/python3/dist-packages/blueman/plugins/applet/DiscvManager.py:123)
warning: Adapter is None 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.PPPSupport.PPPSupport'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ExitItem.ExitItem'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Networking.Networking'> 
_________
load_nap_settings (/usr/lib/python3/dist-packages/blueman/plugins/applet/Networking.py:36)
Loading NAP settings 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AppIndicator.AppIndicator'> 
_________
[... 55 lines cut ...]
_________
on_adapter_property_changed (/usr/lib/python3/dist-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Class 1966348 
_________
on_adapter_property_changed (/usr/lib/python3/dist-packages/blueman/plugins/applet/DiscvManager.py:96)
prop UUIDs dbus.Array([dbus.String('00001112-0000-1000-8000-00805f9b34fb'), dbus.String('00001801-0000-1000-8000-00805f9b34fb'), dbus.String('0000110e-0000-1000-8000-00805f9b34fb'), dbus.String('00001106-0000-1000-8000-00805f9b34fb'), dbus.String('00001800-0000-1000-8000-00805f9b34fb'), dbus.String('00001105-0000-1000-8000-00805f9b34fb'), dbus.String('00001200-0000-1000-8000-00805f9b34fb'), dbus.String('0000110c-0000-1000-8000-00805f9b34fb'), dbus.String('00001104-0000-1000-8000-00805f9b34fb'), dbus.String('0000110a-0000-1000-8000-00805f9b34fb'), dbus.String('0000110b-0000-1000-8000-00805f9b34fb'), dbus.String('00005005-0000-1000-8000-0002ee000001'), dbus.String('00001116-0000-1000-8000-00805f9b34fb'), dbus.String('00001133-0000-1000-8000-00805f9b34fb'), dbus.String('0000112f-0000-1000-8000-00805f9b34fb'), dbus.String('00001132-0000-1000-8000-00805f9b34fb'), dbus.String('00001108-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1) 
_________
RequestPowerState (/usr/lib/python3/dist-packages/blueman/plugins/applet/PowerManager.py:131)
Requesting True 
_________
on_power_state_change_requested (/usr/lib/python3/dist-packages/blueman/plugins/applet/KillSwitch.py:118)
True 
_________
UpdatePowerState (/usr/lib/python3/dist-packages/blueman/plugins/applet/PowerManager.py:159)
off False 
foff False 
on True 
current state True 
new state True 
_________
check (/usr/lib/python3/dist-packages/blueman/plugins/applet/PowerManager.py:119)
callbacks done 
_________
set_adapter_state (/usr/lib/python3/dist-packages/blueman/plugins/applet/PowerManager.py:90)
True

closed 2020-07-27 by stale[bot]