blueman-project/blueman #1263: blueman-2.1.3 fails to start

juanitotc, 2020-05-22:
blueman: 2.1.3
BlueZ: 5.54
Distribution: Tinycorelinux piCore
Desktop environment: flwm

After compiling blueman-2.1.3, blueman-manager ran once, but now times out on startup as per below - blueman-applet starts, pauses to wait for blueman-manager, but the two don't seem to connect.

networkmanager is not managing the network connection.

I believe appindicator is optional, are there any other troubleshooting commands I could try?

```
$ blueman-applet
blueman-applet version 2.1.3 starting
blueman-applet 12.14.42 ERROR    PluginManager:90 load_plugin: Unable to load plugin module AppIndicator
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/blueman/plugins/applet/AppIndicator.py", line 7, in <module>
    require_version('AppIndicator3', '0.1')
  File "/usr/local/lib/python3.8/site-packages/gi/__init__.py", line 129, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace AppIndicator3 not available

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/blueman/main/PluginManager.py", line 88, in load_plugin
    importlib.import_module(self.module_path.__name__ + ".%s" % plugin)
  File "/usr/local/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/local/lib/python3.8/site-packages/blueman/plugins/applet/AppIndicator.py", line 9, in <module>
    raise ImportError("AppIndicator3 not found")
ImportError: AppIndicator3 not found
blueman-applet 12.14.42 WARNING  PluginManager:147 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-tray version 2.1.3 starting

[pause here - start blueman-manager ("starting") nothing happens]

blueman-applet 12.15.07 ERROR    BasePlugin:76 _load     : Error calling StartServiceByName for org.blueman.Mechanism: Failed to activate service 'org.blueman.Mechanism': timed out
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/blueman/main/DBusProxies.py", line 36, in __init__
    self.init()
gi.repository.GLib.GError: g-dbus-error-quark: Error calling StartServiceByName for org.blueman.Mechanism: Failed to activate service 'org.blueman.Mechanism': timed out (20)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/blueman/plugins/BasePlugin.py", line 70, in _load
    self.on_load()
  File "/usr/local/lib/python3.8/site-packages/blueman/plugins/applet/Networking.py", line 24, in on_load
    self.load_nap_settings()
  File "/usr/local/lib/python3.8/site-packages/blueman/plugins/applet/Networking.py", line 45, in load_nap_settings
    m = Mechanism()
  File "/usr/local/lib/python3.8/site-packages/blueman/main/DBusProxies.py", line 15, in __call__
    cls._instance = super().__call__(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/blueman/main/DBusProxies.py", line 43, in __init__
    super().__init__(name='org.blueman.Mechanism', interface_name='org.blueman.Mechanism',
  File "/usr/local/lib/python3.8/site-packages/blueman/main/DBusProxies.py", line 38, in __init__
    raise DBusProxyFailed(e.message)
blueman.main.DBusProxies.DBusProxyFailed: Error calling StartServiceByName for org.blueman.Mechanism: Failed to activate service 'org.blueman.Mechanism': timed out
blueman-applet 12.15.07 ERROR    PluginManager:161 __load_plugin: Failed to load Networking
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/blueman/main/DBusProxies.py", line 36, in __init__
    self.init()
gi.repository.GLib.GError: g-dbus-error-quark: Error calling StartServiceByName for org.blueman.Mechanism: Failed to activate service 'org.blueman.Mechanism': timed out (20)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/blueman/main/PluginManager.py", line 159, in __load_plugin
    inst._load()
  File "/usr/local/lib/python3.8/site-packages/blueman/plugins/AppletPlugin.py", line 57, in _load
    super()._load()
  File "/usr/local/lib/python3.8/site-packages/blueman/plugins/BasePlugin.py", line 70, in _load
    self.on_load()
  File "/usr/local/lib/python3.8/site-packages/blueman/plugins/applet/Networking.py", line 24, in on_load
    self.load_nap_settings()
  File "/usr/local/lib/python3.8/site-packages/blueman/plugins/applet/Networking.py", line 45, in load_nap_settings
    m = Mechanism()
  File "/usr/local/lib/python3.8/site-packages/blueman/main/DBusProxies.py", line 15, in __call__
    cls._instance = super().__call__(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/blueman/main/DBusProxies.py", line 43, in __init__
    super().__init__(name='org.blueman.Mechanism', interface_name='org.blueman.Mechanism',
  File "/usr/local/lib/python3.8/site-packages/blueman/main/DBusProxies.py", line 38, in __init__
    raise DBusProxyFailed(e.message)
blueman.main.DBusProxies.DBusProxyFailed: Error calling StartServiceByName for org.blueman.Mechanism: Failed to activate service 'org.blueman.Mechanism': timed out
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/blueman/main/DBusProxies.py", line 36, in __init__
    self.init()
gi.repository.GLib.Error: g-dbus-error-quark: Error calling StartServiceByName for org.blueman.Mechanism: Failed to activate service 'org.blueman.Mechanism': timed out (20)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/blueman-applet", line 42, in <module>
    BluemanApplet()
  File "/usr/local/lib/python3.8/site-packages/blueman/main/Applet.py", line 34, in __init__
    self.Plugins.load_plugin()
  File "/usr/local/lib/python3.8/site-packages/blueman/main/PluginManager.py", line 121, in load_plugin
    self.__load_plugin(cls)
  File "/usr/local/lib/python3.8/site-packages/blueman/main/PluginManager.py", line 159, in __load_plugin
    inst._load()
  File "/usr/local/lib/python3.8/site-packages/blueman/plugins/AppletPlugin.py", line 57, in _load
    super()._load()
  File "/usr/local/lib/python3.8/site-packages/blueman/plugins/BasePlugin.py", line 70, in _load
    self.on_load()
  File "/usr/local/lib/python3.8/site-packages/blueman/plugins/applet/Networking.py", line 24, in on_load
    self.load_nap_settings()
  File "/usr/local/lib/python3.8/site-packages/blueman/plugins/applet/Networking.py", line 45, in load_nap_settings
    m = Mechanism()
  File "/usr/local/lib/python3.8/site-packages/blueman/main/DBusProxies.py", line 15, in __call__
    cls._instance = super().__call__(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/blueman/main/DBusProxies.py", line 43, in __init__
    super().__init__(name='org.blueman.Mechanism', interface_name='org.blueman.Mechanism',
  File "/usr/local/lib/python3.8/site-packages/blueman/main/DBusProxies.py", line 38, in __init__
    raise DBusProxyFailed(e.message)
blueman.main.DBusProxies.DBusProxyFailed: Error calling StartServiceByName for org.blueman.Mechanism: Failed to activate service 'org.blueman.Mechanism': timed out
tc@box:~$ Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/blueman/main/DBusProxies.py", line 36, in __init__
    self.init()
gi.repository.GLib.Error: g-io-error-quark: Error calling StartServiceByName for org.blueman.Applet: Timeout was reached (24)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/blueman-tray", line 40, in <module>
    BluemanTray()
  File "/usr/local/lib/python3.8/site-packages/blueman/main/Tray.py", line 16, in __init__
    applet = AppletService()
  File "/usr/local/lib/python3.8/site-packages/blueman/main/DBusProxies.py", line 15, in __call__
    cls._instance = super().__call__(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/blueman/main/DBusProxies.py", line 49, in __init__
    super().__init__(name='org.blueman.Applet', interface_name='org.blueman.Applet',
  File "/usr/local/lib/python3.8/site-packages/blueman/main/DBusProxies.py", line 38, in __init__
    raise DBusProxyFailed(e.message)
blueman.main.DBusProxies.DBusProxyFailed: Error calling StartServiceByName for org.blueman.Applet: Timeout was reached
```

infirit, 2020-05-22:
This is what you can expect when you install into /usr/local. You need to pass the correct options to configure like --prefix=/usr and --sysconfdir=/etc. There may be more you need which may be specific for your distribution.

juanitotc, 2020-05-22:
blueman-2.0 worked fine when compiled to /usr/local and, strangely, blueman-2.1.3 ran once when compiled to /usr/local, just after being installed.

juanitotc, 2020-05-22:
I believe I found the issue, but not how to fix it.

This distro uses squashfs packages loop mounted and symlinked to the root filesystem.

When I installed blueman it worked when it was actual files in the root filesystem - when I use a squashfs package for blueman, it does not start, this was confirmed by over-writing the symlinks with the actual files, which made things work again.

Is there a reason why blueman-2.1.3 might not follow symlinks (blueman-2.0 does not have this problem)?

juanitotc, 2020-05-22:
I found the problem - /usr/local/etc/dbus-1/system.d/org.blueman.Mechanism.conf needs to be a real file and not a symlink.

Given that this is a non-system.d distro that is a little surprising...

infirit, 2020-05-22:
That file is not related to systemd the naming may have confused you. We use an **autostartable** dbus service that handles some low level setup. This setup requires root so we don't have it run all the time.

For dbus to be aware and start the necessary binary when we access the name (org.blueman.Mechanism) on dbus we need to install that conf in the right place. Permission may also be a factor but I don't remember.

infirit, 2020-05-22:
Sorry, that may have been more confusing.

**org.blueman.Mechanism.service** and **org.blueman.Applet.service**
These dbus service files are required so we can autostart blueman-mechanism and blueman-applet when we access their names on the dbus.

**org.blueman.Mechanism.conf**
This file sets up some dbus permission what and who is allowed to access the org.blueman.Mechanism dbus service.

**org.blueman.policy.in**
Lastly we use policykit to to allow non root users to access org.blueman.Mechanism.

So it is important to make sure these files are installed properly in ``/usr``. I think you can indeed run the rest from ``/usr/local``, it is however I generally don't recommend as it may break in unsuspecting ways.

juanitotc, 2020-05-22:
..but the problem is not /usr or /usr/local it is that org.blueman.Mechanism.conf needs to be a real file...

juanitotc, 2020-05-24:
I repackaged blueman such that org.blueman.Mechanism.conf is a real file and things work now.

closed 2020-05-24 by cschramm