blueman-project/blueman #1265: Device connects in 'HeadSet' mode, cannot change to "AudioSink'

cghercoias, 2020-05-24:
blueman: version 2.0.8
BlueZ: 5.50-1.2~deb10u1+rpt1
Distribution:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian

Desktop environment: XFCE

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2020-05-24:
What hinders you?

cghercoias, 2020-05-25:
Lol, Thanks for the response. I guess nothing hinders me, other than the
graphical interface.
It does not allow me to change the setting. I tried during pairing with
speaker. Then from the blue-alsa menu (where are three options - headset,
headphones and audiosink).
Only one setting works - headset.
I don't know much about the bluetooth and blueman, it just prompted me to
'report an error' - which I did.
Please let me know what you think and if I should submit any logs. I looked
through dmesg and /var/log/messages and nothing jumps to me.

Many thanks,
*-- Catalin Ghercoias*

cschramm, 2020-05-26:
So when you choose A2DP sink in blueman it just sticks to HSP? Sounds like a pulseaudio-related issue. Not sure how to debug.

Sounds a bit like #1242 #1209 #1031 so you could give BlueZ 5.51+ a try.

closed 2020-08-02 by github-actions[bot]