blueman-project/blueman #1266: Paired but failed to connect

Baptistouta, 2020-05-24:
blueman:2.0.5
BlueZ:5
Distribution:lubuntu 18.04
Desktop environment:lxde

I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

My audio devices are paired but I cannot connect them. When I make synchronie audio blueman gives me 
> blueman.bluez.errors.DBusFailedError: Protocol not available... 

I used blueman-applet, please find what [it returned](https://pastebin.com/TfYRpfyM) 
Futhermore, I tried to follow https://github.com/blueman-project/blueman/wiki/Troubleshooting 
> A2DP audio device PAIRED but fails to connect.

Finally I couldn't because with LXDE, the directory /var/lib/gdm doesn't exist. It seems to be replaced by lightdm , but we are not allowed to execute or read or write this directory.

Thank You for helping me

cschramm, 2020-05-24:
`pulseaudio-module-bluetooth` installed?

Baptistouta, 2020-05-24:
No, it wasn't. I installed and it solved my problem. Thank you very much.

closed 2020-05-24 by Baptistouta