blueman-project/blueman #1269: blueman-applet 2.1.3 open popup window in background or minimized

joakim-tjernlund, 2020-05-28:
blueman: 2.1.3
BlueZ: git master
Distribution: Gentoo 
Desktop environment: MATE 1.24

* [ x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Whenever I click on applet to see my devices, the window either opens behind other windows or
even minimized.

I ran a blueman-applet --loglevel debug and ther was no output when a clicked the icon

joakim-tjernlund, 2020-05-28:
a blueman-manager --loglevel debug will open window on top though.

cschramm, 2020-05-29:
I've seen that as well recently. I'm pretty sure it's related to MATE 1.24 somehow. :thinking:

infirit, 2020-05-29:
This is because of GAppinfo but I am not sure how to fix this. It is definitely the focus stealing prevention most metacity based windowmanager do. I tried to drop the event time and a few other things, no joy 😞. 

I can workaround this by using a GDesktopAppinfo but then we need to provide desktop files for all blueman launch scripts. Adding desktop files is the easiest. The question is which do we want to show in the desktop menu.

joakim-tjernlund, 2020-05-29:
you mean:
```
cat /etc/xdg/autostart/blueman.desktop
[Desktop Entry]
Name=Blueman Applet
Comment=Blueman Bluetooth Manager
Icon=blueman
Exec=blueman-applet
Terminal=false
Type=Application
Categories=
```
What should be in there ?

joakim-tjernlund, 2020-05-29:
Why is blueman special here? its the only app I have with this problem?

cschramm, 2020-05-30:
@infirit: Doesn't seem to be GAppInfo alone. If I just run

```
from gi.repository import Gio
Gio.AppInfo.create_from_commandline("blueman-manager", "blueman", Gio.AppInfoCreateFlags.SUPPORTS_STARTUP_NOTIFICATION).launch()
```

which is basically the same just not from within blueman-applet, the window opens on top.

I do have similar issues with other applications in MATE, though, e.g. Thunderbird windows opening in background when clicking `mailto:` links.

infirit, 2020-05-30:
Spend some time poking at things. The reason the focus stealing logic kicks in is because ``Gtk.get_current_event_time()`` is always 0 (zero). Combine this with Gdk.AppLaunchContext requiring a proper  timestamp for it to give applications focus and we now know why it is happening.

But I hear you say, isn't ``Gtk.get_current_event_time`` to fallback to Gdk.CURRENT_TIME?? Yes it does but we don't execute ui code in blueman-applet anymore so there is no timestamp. In fact we can replace ``Gtk.main()`` with ``GLib.MainLoop().run()`` and all works as expected.

The best solution I can think of is the use the time module to retrieve a timestamp. Patch below for people to test. I'll open a PR when I am satisfied this is working as it should.

```patch
From d2054e899aae08f898f439eb5789e79178cd879d Mon Sep 17 00:00:00 2001
From: Sander Sweers <infirit@gmail.com>
Date: Sat, 30 May 2020 13:00:21 +0200
Subject: [PATCH] Use time module to get a timestamp

We can not use Gtk.get_current_event_time as we do not execute any ui
code anymore and it always returns 0 (zero).
---
 blueman/Functions.py | 3 ++-
 1 file changed, 2 insertions(+), 1 deletion(-)

diff --git a/blueman/Functions.py b/blueman/Functions.py
index 8b7626c3..ed27248a 100644
--- a/blueman/Functions.py
+++ b/blueman/Functions.py
@@ -36,6 +36,7 @@ import fcntl
 import struct
 import socket
 import array
+import time
 
 import cairo
 
@@ -114,7 +115,7 @@ def launch(
 ) -> bool:
     """Launch a gui app with starup notification"""
     display = Gdk.Display.get_default()
-    timestamp = Gtk.get_current_event_time()
+    timestamp = int(time.clock_gettime(time.CLOCK_MONOTONIC_RAW))
     context = display.get_app_launch_context()
     context.set_timestamp(timestamp)
     if sn:
-- 
2.26.2
```

joakim-tjernlund, 2020-05-31:
Tried that patch on 2.1.3 but it did not apply?

joakim-tjernlund, 2020-05-31:
I handedited this change into 2.1.3 but now the window is always minimized

cschramm, 2020-05-31:
:thinking: I can confirm that `Gtk.get_current_event_time()` is `0` while `int(time.clock_gettime(time.CLOCK_MONOTONIC_RAW))` is not and that the issue is related to the context we pass as it does not happen if I remove the argument but the patch does not help in my case.

cschramm, 2020-06-03:
I played with my pacman cache a little bit and concluded that this is *not* related to MATE 1.24. I'm still sure, though, that it started at some point, maybe MATE 1.22 but I did not pay too much attention as my systems are weird anyway... :confused:

infirit, 2020-06-05:
See #1272. This works in all situation for me now.

joakim-tjernlund, 2020-06-05:
Ahh, but would you mind create a minimal patch for stable as well?

infirit, 2020-06-05:
@joakim-tjernlund it is in master now so you can use ``blueman-9999``. Or put https://github.com/blueman-project/blueman/pull/1273/commits/f9f0e7619d4e46cb6a42ae47b9a44324839f8779 in ``/etc/portage/patches/net-wireless/blueman``

joakim-tjernlund, 2020-06-05:
Thanks, I will test when I get home(during the weekend that is)

joakim-tjernlund, 2020-06-07:
Works nicely, thanks :)

closed 2020-06-08 by cschramm