blueman-project/blueman #1274: 2.1.2 Cannot pair Logitech K860 keyboard

lk-chen, 2020-06-06:
blueman: 2.1.2
BlueZ: 5.50
Distribution: Debian
Desktop environment: X-Cinnamon

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I am able to find and connect to the device, but failed pairing with it.
```
blueman-manager 22.25.53 ERROR    Manager:201 error_handler:  Authentication Canceled
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/bluez/Base.py", line 81, in callback
    value = proxy.call_finish(result).unpack()
gi.repository.GLib.GError: g-io-error-quark: GDBus.Error:org.bluez.Error.AuthenticationCanceled: Authentication Canceled (36)
```

cschramm, 2020-06-06:
Please provide output from `blueman-applet --loglevel debug` of the failed pairing and also check https://github.com/blueman-project/blueman/wiki/Troubleshooting#pairing-fails-on-the-remote-devices-side-wihout-a-pairing-dialog-showing.

lk-chen, 2020-06-07:
Finally I paired with logitech provided universal receiver. Closing this issue.

closed 2020-06-07 by lk-chen