blueman-project/blueman #1275: Bluetooth does not add any device, wether audio device or my samsung galaxy J7

descholar-ceo, 2020-06-06:
blueman: 2.0.5
BlueZ: blueman-applet
Distribution: Linux mint 19.3 Tricia
Desktop environment: Cinnamon desktop

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2020-06-07:
Please provide output from `blueman-applet` (see the troubleshooting page) of the failing process.

descholar-ceo, 2020-06-07:
@cschramm It keeps searching and it cannot find any bluetooth device

cschramm, 2020-06-11:
Did you try searching in `bluetoothctl`? If nothing shows up but there are discoverable devices, that's probably an issue with the driver / firmware / controller.

descholar-ceo, 2020-06-12:
This is what I get when I run `show`
[image]

and when I run `scan on` I get this
[image]

When I run `devices` It shows nothing

===>It seems like I am not getting any device, (I have my samsung galaxy J7 bluetooth on and visible to all near by devices on) and is able to be paired.

cschramm, 2020-06-12:
That pretty much confirms "driver / firmware / controller". Nothing we can do from the blueman side.

Check your system logs for errors on start and when searching. The driver or bluetoothd might give some hint. Also find your chipset with `lsusb` or `lspci` and see if you can find similar issues with it on the Internets.

closed 2020-06-12 by cschramm