blueman-project/blueman #1276: Headset connects but don't appear in Volume Control e.i. no sound

DaVinci1987, 2020-06-09:
Hi there,
Device appears as new, is added as Headset and connects but don't show in Volume Control as Output device. 
This device worked with the buster distro.
I tried Twister OS and it don't work.

Thanks for your help in advance
attached is a screenshot

[url=https://i.imgur.com/a6ZyUTD.png]
  [img]https://imgur.com/a6ZyUTDl.png[/img]
[/url]

cschramm, 2020-06-11:
Sounds like a pulseaudio issue (its logs might show something).

Please right-click on the device in blueman and open the info dialog to provide the UUIDs and, just in case, check that you have the bluetooth pulsaudio module.

DaVinci1987, 2020-06-14:
Hi cschramm, 
maybe I'm blind but I cannot find this info dialog. I've tried another bluetooth headset but same issue.
How do I check whether I have pulsaudio module or not? 

Thank you in advance
[url=https://i.imgur.com/ZD5FYKc.png]
  [img]https://imgur.com/ZD5FYKc.png[/img]
[/url]

DaVinci1987, 2020-06-14:
sudo apt-get install pulseaudio pulseaudio-module-bluetooth

pulseaudio-module-bluetooth is already the newest version (12.2-4+deb10u1)

cschramm, 2020-06-14:
Maybe you did not provide the basic information asked for in the issue template so that I don't even have a clue what verison of blueman you're using.

You can get the UUIDs with `bluethoothcl info` (assuming a certain version of BlueZ) as well if blueman does not have the info dialog.

DaVinci1987, 2020-06-14:
Sry for not providing all information.
I bought a new pi 4 and installed TwisterOS. All I want is a working speaker setup :)
After `bluetoothctl info` command, this appears:

Device E8:07:BF:A8:93:9C (public)
	Name: SoundBuds Flow
	Alias: SoundBuds Flow
	Class: 0x00240404
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Vendor specific           (00000000-0000-0000-0099-aabbccddeeff)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)

cschramm, 2020-06-14:
Looks like a proper audio sink that pulseaudio should pick up. Nothing we could do there. Maybe pulseaudio gives a hint in some log entries.

closed 2020-06-14 by cschramm