blueman-project/blueman #1278: blueman-applet tooltip shows "Bluetooth Enabled" even with bluetooth disabled

fdmarcin, 2020-06-15:
blueman: 2.1.3-1
BlueZ: 5.54-2
Distribution: Manjaro
Desktop environment: XFCE 4.14.2

`blueman-applet` tooltip always says "Bluetooth Enabled".

The string `"Bluetooth Enabled"` appears in [ShowConnected.py](https://github.com/blueman-project/blueman/blob/master/blueman/plugins/applet/ShowConnected.py#L60) and [StatusIcon.py](https://github.com/blueman-project/blueman/blob/master/blueman/plugins/applet/StatusIcon.py#L32) and all the `.po` files.

[image]

When running `blueman-applet` from terminal (log below), the error complains about missing `AppIndicator3`, so this might be my fault. All my status indicators have worked correctly.

<details><summary>Log: starting <code>blueman-applet</code> from terminal</summary>

```shell
blueman-applet version 2.1.3 starting
Stale PID, overwriting
blueman-applet 15.46.57 ERROR    PluginManager:90 load_plugin: Unable to load plugin module AppIndicator
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/blueman/plugins/applet/AppIndicator.py", line 7, in <module>
    require_version('AppIndicator3', '0.1')
  File "/usr/lib/python3.8/site-packages/gi/__init__.py", line 129, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace AppIndicator3 not available

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/blueman/main/PluginManager.py", line 88, in load_plugin
    importlib.import_module(self.module_path.__name__ + ".%s" % plugin)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3.8/site-packages/blueman/plugins/applet/AppIndicator.py", line 9, in <module>
    raise ImportError("AppIndicator3 not found")
ImportError: AppIndicator3 not found
blueman-applet 15.46.57 WARNING  PluginManager:147 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 15.46.58 WARNING  PluginManager:147 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-tray version 2.1.3 starting
blueman-applet 15.46.58 WARNING  DiscvManager:109 update_menuitems: warning: Adapter is None
```
</details>

After I installed `libappindicator-gtk3`, blueman-applet isn't showing any tooltip now :see_no_evil: 

<details><summary>Log: starting <code>blueman-applet</code> from terminal after installing <code>libappindicator-gtk3</code></summary>

```shell
blueman-applet version 2.1.3 starting
blueman-applet 16.04.53 WARNING  PluginManager:147 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 16.04.53 WARNING  PluginManager:147 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-tray version 2.1.3 starting
Stale PID, overwriting
blueman-tray version 2.1.3 starting
There is an instance already running
blueman-applet 16.04.53 WARNING  DiscvManager:109 update_menuitems: warning: Adapter is None

(blueman-tray:53054): Gdk-CRITICAL **: 16:04:53.971: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed
blueman-manager version 2.1.3 starting
```
</details>

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2020-06-18:
Please provide the output of starting blueman-applet with `--loglevel debug`.

fdmarcin, 2020-06-18:
Here you go! After trying it with Appindicator3, I uninstalled the `libappindicator-gtk3` package.

<details><summary><code>blueman-applet --loglevel debug</code></summary>

```
blueman-applet version 2.1.3 starting
Stale PID, overwriting
blueman-applet 11.59.06 INFO     PluginManager:85 load_plugin: ['KillSwitch', 'DhcpClient', 'Menu', 'AppIndicator', 'AuthAgent', 'TransferService', 'GameControllerWakelock', 'StandardItems', 'PowerManager', 'NMDUNSupport', 'RecentConns', 'Networking', 'NetUsage', 'DiscvManager', 'ExitItem', 'ShowConnected', 'StatusIcon', 'DBusService', 'SerialManager', 'NMPANSupport', 'PPPSupport']
blueman-applet 11.59.06 ERROR    PluginManager:90 load_plugin: Unable to load plugin module AppIndicator
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/blueman/plugins/applet/AppIndicator.py", line 7, in <module>
    require_version('AppIndicator3', '0.1')
  File "/usr/lib/python3.8/site-packages/gi/__init__.py", line 129, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace AppIndicator3 not available

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/blueman/main/PluginManager.py", line 88, in load_plugin
    importlib.import_module(self.module_path.__name__ + ".%s" % plugin)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3.8/site-packages/blueman/plugins/applet/AppIndicator.py", line 9, in <module>
    raise ImportError("AppIndicator3 not found")
ImportError: AppIndicator3 not found
blueman-applet 11.59.06 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 11.59.06 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 11.59.06 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 11.59.06 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 11.59.06 WARNING  PluginManager:147 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 11.59.06 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 11.59.06 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 11.59.06 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 11.59.06 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 11.59.06 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 11.59.06 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 11.59.06 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 11.59.06 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 11.59.06 INFO     Networking:31 load_nap_settings: Loading NAP settings
blueman-applet 11.59.06 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 11.59.06 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 11.59.06 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 11.59.06 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 11.59.06 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 11.59.06 WARNING  PluginManager:147 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 11.59.06 INFO     KillSwitch:106 io_event  : killswitch registered 1
blueman-applet 11.59.06 INFO     KillSwitch:122 io_event  : State: False
blueman-applet 11.59.06 INFO     PowerManager:171 update_power_state: off True | foff False | on True | current state True | new state False
blueman-applet 11.59.06 INFO     PowerManager:175 update_power_state: Signalling False
blueman-tray version 2.1.3 starting
blueman-applet 11.59.06 INFO     KillSwitch:87 _on_connman_vanished: net.connman vanished
blueman-applet 11.59.06 INFO     TransferService:210 _on_dbus_name_appeared: org.bluez.obex :1.67
blueman-applet 11.59.06 INFO     Applet:51 _on_dbus_name_appeared: org.bluez :1.23
blueman-applet 11.59.06 INFO     BluezAgent:97 register_agent: Register Agent
blueman-applet 11.59.06 INFO     Networking:67 set_nap   : set nap False
blueman-applet 11.59.06 WARNING  DiscvManager:109 update_menuitems: warning: Adapter is None
blueman-applet 11.59.06 INFO     AgentManager:15 on_registered: /org/bluez/obex/agent/blueman
blueman-applet 11.59.06 INFO     ShowConnected:38 enumerate_connections: Found 0 existing connections
```
</details>

infirit, 2020-06-20:
This looks like copy/paste bug but..

https://github.com/blueman-project/blueman/blob/2-1-stable/blueman/plugins/applet/ShowConnected.py#L61

Below may solve the problem so please try it out.

```diff
@@ -58,7 +58,7 @@ def update_statusicon(self):
            if status:
                self.parent.Plugins.StatusIcon.set_text_line(0, _("Bluetooth Enabled"))
            else:
-               self.parent.Plugins.StatusIcon.set_text_line(0, _("Bluetooth Enabled"))
+               self.parent.Plugins.StatusIcon.set_text_line(0, _("Bluetooth Disabled"))

    def on_manager_state_changed(self, state):
        if state:
```

fdmarcin, 2020-06-22:
@infirit Hi!

I found this file in `/usr/lib/python3.8/site-packages/blueman/plugins/applet` and changed this line.

It seems like it didn't work: when I switch Bluetooth on and off using the applet I can see the `Bluetooth Enabled` tooltip on both states.

A few times, a second after disabling it, **I could see "Disabled" for a short moment** only for the tooltip to change to "Enabled" again while the state was still disabled.

closed 2020-08-24 by infirit