blueman-project/blueman #1281: Disconnecting headset "No adapter found"

callegula, 2020-06-17:
blueman:2.1.3
BlueZ:
Distribution: fedora 
Desktop environment: i3
versions as of june17 2020

blueman-manager 21.06.35 ERROR    Manager:118 on_dbus_name_appeared: Default adapter not found, trying first available.
blueman-manager 21.06.35 ERROR    Manager:122 on_dbus_name_appeared: No adapter(s) found, exiting

Happens time and time again.

Disconnecting headset

cschramm, 2020-06-18:
How do you disconnect? On your computer or from the device itself?

If I get you right, it's unpredictable when it happens, correct?

callegula, 2020-06-18:
Hi 

I disconnect 90% by powering of the headphones. 

Regards 

 

Links:
------
[1]
https://github.com/blueman-project/blueman/issues/1281#issuecomment-645805682
[2]
https://github.com/

BraveJero, 2020-08-18:
The same happens to me. Random disconnects

closed 2020-10-25 by github-actions[bot]