blueman-project/blueman #1288: No Bluetooth found plug in a dongle to use the Bluetooth

iffishells, 2020-07-03:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

glaubergf, 2020-07-20:
OS Debian 10 Xfce.
**************************
All dependency packages installed as referenced on the Debian website. Blueman opens more when choosing to configure a new device, a window opens, locks and closes without any error return.
**************************
Command lsusb:

Bus 002 Device 002: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
**************************
Command sudo dmesg | grep idProduct=0001:

[132394.496402] usb 2-2: New USB device found, idVendor=0a12, idProduct=0001, bcdDevice=25.20
**************************
In Devices, searching for device returns this error.

Command blueman-assistant:

/usr/bin/blueman-assistant:22: DeprecationWarning: Importing dbus.glib to use the GLib main loop with dbus-python is deprecated.
Instead, use this sequence:

    from dbus.mainloop.glib import DBusGMainLoop

    DBusGMainLoop(set_as_default=True)

  import dbus.glib
_________
get_interface_version (/usr/lib/python3/dist-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
/usr/bin/blueman-assistant:80: DeprecationWarning: Gtk.Assistant.set_page_header_image is deprecated
  self.assistant.set_page_header_image(pages[PAGE_INTRO], get_icon("blueman", 32))
_________
SetAdapter (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:271)
None 
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/bluez/errors.py", line 146, in warp
    return func(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/blueman/bluez/Adapter.py", line 135, in start_discovery
    self.get_interface().StartDiscovery()
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.bluez.Error.InProgress: Operation already in progress

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/blueman-assistant", line 311, in <module>
    Assistant()
  File "/usr/bin/blueman-assistant", line 154, in __init__
    self.dev_widget.List.DiscoverDevices()
  File "/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py", line 401, in DiscoverDevices
    self.Adapter.start_discovery()
  File "/usr/lib/python3/dist-packages/blueman/bluez/errors.py", line 148, in warp
    raise parse_dbus_error(exception)
blueman.bluez.errors.DBusInProgressError: Operation already in progress

cschramm, 2020-08-01:
@glaubergf: Are you providing information on @iffishells' problem? It neither looks like no adapter is available nor did I get what the actual problem is. There are some deprecation warnings and non-fatal errors but nothing that should cause any actual trouble.

closed 2020-08-01 by cschramm