blueman-project/blueman #1292: Misleading UI message "Failed: Connection in progress"

zoot, 2020-07-13:
blueman-applet: 2.0.5
BlueZ:
Distribution: Ubuntu 18.04
Desktop environment: XFCE

When connecting a device, after a reboot, or even resuming from suspend, such as a keyboard, then accessing the context menu "recent connections", to select the device, a notification message shows with an emphasis on failure rather than notifying the user that the connection process is underway.  So, from a user perspective, one assumes there has been a failure, rather than progress waiting.

What shows:

Failed:
connection in progress ...

What should show:

Busy:
connection in progress ...

cschramm, 2020-07-13:
As there is no such notification from blueman, I guess you're seeing an actual error from BlueZ that tries to tell you that there's already an operation in progress and thus connecting is not possible. However, there's no such message as "connection in progress" from BlueZ either. It should probably be either just "In Progress" (device related) or "Operation already in progress" (adapter related).

If that happens every time you use recent connections, probably somehow the connection attempt is started twice. Could you please provide the blueman-applet output for that process so that we can have a look?

zoot, 2020-07-14:
Thanks for rapidly responding @cschramm 

You're right, I was posting from (flawed) memory, sorry. Here are the screen shots. 

[image]
[image]

The keyboard is always on and the broadcast icon flashing when this occurs.

Your suggested alternative wording would be far better than "Failed to connect: In progress", with the emphasis switched around.

Here's the log output ...

[blueman-applet.log](https://github.com/blueman-project/blueman/files/4917611/blueman-applet.log)

cschramm, 2020-08-29:
There does not seem to be even a single connection attempt in the log.

closed 2020-09-13 by github-actions[bot]