blueman-project/blueman #1296: blueman-applet occasionally launches 2 instances of blueman-tray

j-piecuch, 2020-07-18:
blueman: 2.1.3
BlueZ: 5.54
Distribution: Manjaro 20.0.3
Desktop environment: i3

I think this is a race condition involving the lock file.
If the `StatusIcon` and `AppIndicator` plugins are both enabled, then during startup the `StatusIcon` plugin will launch `blueman-tray` twice in `_on_plugins_changed()`: once when `StatusIcon` is loaded, once when `AppIndicator` is loaded. The function checks if there is an instance already running by reading the tray's lock file, but that's racy, since the tray can be running without having created the lock file. There's yet another check in `blueman-tray` using `check_single_instance()`, but that's racy too, here's how both trays can pass the check, assuming a stale lock file exists:
1. Tray 1 sees that a lock file exists, then reads a stale PID.
2. Tray 2 sees that a lock file exists, reads the same stale PID, removes the lock file, and writes its own lock file.
3. Tray 1 still thinks that the lock file is stale, so it removes tray 2's lock file and writes its own lock file.

Overall, there seem to be two problems here, and both need to be addressed for this to work correctly:
1. The check for a running tray in `_on_plugins_changed()` inside `StatusIcon`: even if `check_single_instance()` worked correctly, we could end up with the wrong instance of the tray running (i.e. the one launched first). This could be fixed by storing the PID of the tray process in a field of the `StatusIcon` class instead of reading it from the lock file, and reading the lock file when initializing that field in `StatusIcons`'s `__init__()` to handle the case where `blueman-tray` is launched first.
2. `check_single_instance()`: I think it can be fixed by serializing calls that check the same lock file. We can use [fcntl.flock](https://docs.python.org/3/library/fcntl.html#fcntl.flock) for that.

If my explanation is correct, then a workaround would be to disable the `AppIndicator` plugin on systems that can display GTK status icons.

infirit, 2020-07-27:
See, #1304

cschramm, 2020-08-01:
@j-piecuch: Can you check current master which uses Gio / Gtk Application?

j-piecuch, 2020-08-01:
I have tested current master with GNOME 3.30, which is where I encountered this issue most frequently, and the issue with launching 2 instances seems to be fixed. The issue with the wrong instance of the tray running also appears to have been resolved.
Thanks!

closed 2020-08-01 by j-piecuch