blueman-project/blueman #1297: Refresh

silocoder, 2020-07-20:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2020-07-20:
Do you use the AppIndicator plugin? Is it the same if you disable it? What do you mean with "there is a scan for devices"? Running the discovery from blueman-manager?

Please provide the blueman-applet output of that process. Also, a short screencast might be helpful.

silocoder, 2020-07-20:
I think there may have been a conflict with the other bluetooth applet that comes with Linuxmint.  I removed it and don't seem to have that problem. I will see. Thanks

closed 2020-07-20 by cschramm