blueman-project/blueman #1303: Need help debugging my Bluetooth speaker

silocoder, 2020-07-24:
LM 20

The speaker has been a problem with using on Linux. Not sure why, sometimes it connects other times it doesn't. I have remove, rescan and do all sorts of power on off hoops.  Never had problem with it on the Mac. Would appreciate some advise how to debug it and figure out where/what to look.

Here is info I get from command line. Thanks!

```
~$ bluetoothctl show
Controller 80:56:F2:A5:BA:0C (public)
	Name: silocoder-GT60-2PC
	Alias: silocoder-GT60-2PC
	Class: 0x001c010c
	Powered: yes
	Discoverable: no
	DiscoverableTimeout: 0x000000b4
	Pairable: yes
	UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
	UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
	UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
	UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0535
	Discovering: no
Advertising Features:
	ActiveInstances: 0x00
	SupportedInstances: 0x05
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name

~$ bluetoothctl devices
Device DC:8C:EC:A2:1E:08 ZoeeTree S1

~$ bluetoothctl info DC:8C:EC:A2:1E:08
Device DC:8C:EC:A2:1E:08 (public)
	Name: ZoeeTree S1
	Alias: ZoeeTree S1
	Class: 0x00340404
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)

~$ bluetoothctl connect DC:8C:EC:A2:1E:08
Attempting to connect to DC:8C:EC:A2:1E:08
[CHG] Controller 80:56:F2:A5:BA:0C Class: 0x00000000
Failed to connect: org.bluez.Error.Failed
```

After trying all sorts of variations I got it to connect with this. But Why it worked? Did not even power recycle the speaker.

```
~$ bluetoothctl scan on
Discovery started
[CHG] Controller 80:56:F2:A5:BA:0C Discovering: yes
[NEW] Device 6B:4D:0D:6F:EC:71 6B-4D-0D-6F-EC-71
[DEL] Device DC:8C:EC:A2:1E:08 ZoeeTree S1
[CHG] Device 6B:4D:0D:6F:EC:71 TxPower is nil
[CHG] Device 6B:4D:0D:6F:EC:71 RSSI is nil
[DEL] Device 6B:4D:0D:6F:EC:71 6B-4D-0D-6F-EC-71
[CHG] Controller 80:56:F2:A5:BA:0C Class: 0x00000000
[CHG] Controller 80:56:F2:A5:BA:0C Powered: no
[CHG] Controller 80:56:F2:A5:BA:0C Discovering: no
[DEL] Controller 80:56:F2:A5:BA:0C silocoder-GT60-2PC [default]
~$ bluetoothctl devices
Device DC:8C:EC:A2:1E:08 ZoeeTree S1
~$ bluetoothctl devices
Device DC:8C:EC:A2:1E:08 ZoeeTree S1
~$ bluetoothctl connect DC:8C:EC:A2:1E:08
Attempting to connect to DC:8C:EC:A2:1E:08
[CHG] Device DC:8C:EC:A2:1E:08 Connected: yes
Connection successful
```

infirit, 2020-07-24:
Just to be clear, we are not the people who maintain BlueZ which tools you are using. We are a frontend for BlueZ for the desktop.

But... Do you dual-boot between mac and linux? If so do you use the same speaker?

silocoder, 2020-07-24:
I understand but No dual boot mac. Is there a BlueZ forum I can pose issues on?

cschramm, 2020-08-29:
There are mailing lists and IRC channels, see http://www.bluez.org/contact/.

closed 2020-11-06 by github-actions[bot]