blueman-project/blueman #1310: Auto-receive files from trusted devices is broken

piotrgazda75, 2020-07-30:
blueman: 2.1.2-1
BlueZ: 5.53-0ubuntu3
Distribution: Mint 20 amd64
Desktop environment: MATE 1.24.0

Hello!

I have my HUAWEI P20 Pro paired and trusted and every time I send to my MSI GP73 laptop one or few files I have to click "Accept" for EVERY single file. This is definitely not right :-(

Best regards
Peter

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2020-07-30:
There should be an "Always accept" option next to "Accept" that enables such an "auto-receive". Trusting a device has nothing to do with that.

piotrgazda75, 2020-08-05:
Hello!

Looks like they forgot about it. Please, check the screenshot in
attachment. There are only "Accept" and "Reject" options. I think
MATE's devs should be informed.

Best regards
Piotr Gazda

On Thu, 30 Jul 2020 11:35:07 -0700
Christopher Schramm <notifications@github.com> wrote:

> There should be an "Always accept" option next to "Accept" that
> enables such an "auto-receive". Trusting a device has nothing to do
> with that.
>

cschramm, 2020-08-05:
[image]

Might be a shortcoming of your notification daemon. Not sure right now. Will check the code.

cschramm, 2020-08-05:
Looks like I was completely wrong about this, sorry. ("Always accept" exists for incoming service connections, not for incoming files, and sets the device to trusted.) There is a configuration to make it work as you expect. From the tray icon's context menu open "Local Services", and tick "Accept files from trusted devices" for the "Transfer" service.

closed 2020-10-12 by github-actions[bot]