blueman-project/blueman #1312: linuxlite 5 using bluetooth manager wont connect to my apple wireless keyboard A1016

maclinuxlite, 2020-07-31:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I tried and tried :) seem it really cant connect. 
I was able to search and apple wireless keyboard appears. then clicked pair but nothing happends and it will give pairing failed.
if i use setup. same. but i can press the apple wireless keyboard return key and i will get a pop up alert that gives a PIN number.
not dialog box. so i tried to enter the PIN on my wired keyboard with enter. nothing  happens. if i press apple wireless keyboard return key. it will just give another pop up PIN number. 

Thank You

cschramm, 2020-07-31:
I'd like to tell you to check the manufacturer's manual on how to pair your device but we all know how good that'd work out for that manufacturer. :stuck_out_tongue_winking_eye: 

You have to enter the code blueman shows you on the keyboard that you're trying to pair, I think finalized with the return key.

infirit, 2020-09-08:
I don't have a keyboard that behaves like that. Is there a affordable one I can buy?

Follow https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-blueman and start blueman-applet from a terminal. Then initiate the pair from blueman-manager and post the full log here.

cschramm, 2020-09-09:
Actually we're not using BlueZ' full potential there as `org.bluez.Agent1.DisplayPasskey` gets passed the number of (correctly) entered digits that we could use for a more intuitive presentation but our agent just ignores that. Also, we might display an incorrect PIN code < 100000 as we do not zero-pad to six digits.

closed 2020-09-24 by github-actions[bot]