blueman-project/blueman #1320: RFC: Turn i18n on by default in DeviceClas

infirit, 2020-08-01:
The one place blueman doesn't do a good job wrt to localization is the major device class. My suggestion is to just make everything translatable and always use the localized string like we do in the rest of the ui.

Any objections or concerns?

cschramm, 2020-08-01:
With "doesn't do a good job" you mean it is not localized at all, opposed to minor device classes, right? Well, we display it only if the minor class is "uncategorized", but we do display it after all, so it would make sense to localize it, yes.

cschramm, 2020-08-01:
Regarding the actual translations, I always think about providing some help to translators, mentioning https://www.bluetooth.com/de/specifications/assigned-numbers/baseband/ and https://www.bluetooth.com/de/specifications/assigned-numbers/service-discovery/ (those pages even have some (strange) translations, based on the user agent language), maybe also http://git.kernel.org/cgit/bluetooth/bluez.git/tree/lib/sdp.h. I'm not sure how to place that without flooding our code with comments. A wiki page / Markdown file would be nice, but we need to direct people to find it. Also, the URLs and pages of the Bluetooth SIG tend to change from time to time.

infirit, 2020-08-01:
>With "doesn't do a good job" you mean it is not localized at all, opposed to minor device classes, right?

Yes but not only for the major ones. The GATT and a couple minor classes also do not handle localization. I'll open a PR making all strings translatable.

I also do not have any good suggestions how to help translator.. I do know not many will actually read the source so a wiki entry for translation pointing to SIG and other sources would be better.

closed 2020-08-23 by infirit