blueman-project/blueman #1333: Blueman is not working correctly with linux-swap

marenbraga, 2020-08-08:
blueman: 2.1.2-1
BlueZ:
Distribution:
Desktop environment: linux mint 20 xfce 64

Hi all

i have problems using blueman. when my Mint uses swap memory the connection turns unstable. It's normal?

sorry for my terrible english
bighugs

marina

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

marenbraga, 2020-08-08:
my notebook is a Lenovo Ideapad 330-15IKBR-81FE0002BR

cschramm, 2020-08-09:
That's not expected at all, no. I don't see how swap space could affect blueman, bluetoothd or the bluetooth kernel stuff at all, apart from how it can affect the whole system.

What connection do you refer to an what exactly do you mean when you say that it turns unstable?

marenbraga, 2020-08-09:
Pairing computer with a bluetooth device turns unstable when my swap is in use and my computer enters in sleep mode. it's only happens when it uses swap memory. 

when swap is in use and my note enters in sleeping mode, pairing doesn't works properly and i need to restart all my system to fix it.

marenbraga, 2020-08-09:
ideias?

cschramm, 2020-08-09:
That makes kinda sense. There is suspend (suspend to RAM) and hibernate (suspend to disk). Hibernate requires swap space, so I guess you use hibernate if swap is space is available and that's when you encounter the problems. It's typically related to your specific driver, however, you did not mention what happens if you try to pair after hibernation. To temporarily fix it, you could try to reload the respective kernel module. Restarting bluetoothd might also work, but I guess not. To permanently fix it, you'd probably need a newer driver, possibly not even available.

marenbraga, 2020-08-09:
Thanks Christopher. It's exactly what's going on. After hibernation my device turns unable to pairing. It's visible, but unable to pairing. I try turn my device (sound box) off and on again, but still unable to pairing. The same occurs when i restart my bluetooth. I need to restart my note for my device works.

i will hope for a new kernel then.

thanks bro

closed 2020-08-10 by cschramm