blueman-project/blueman #1335: Bus Error blueman-applet crash on NAP connection

infirit, 2020-08-09:
I am getting a Bus Error after trying to connect a NAP and blueman-applet dies. Can you try to connect, does not have to succeed, a NAP connection on a phone that supports it? I am getting this as far back as 833a8085a5885403f0f59dac72ea3cedd8f05ced so it is probably me but would like confirmation :smile:,

```
blueman-applet 16.49.02 INFO     PowerManager:125 request_power_state: Requesting True
blueman-applet 16.49.02 INFO     KillSwitch:139 on_power_state_change_requested: True
blueman-applet 16.49.02 DEBUG    KillSwitch:152 on_power_state_change_requested: Using mechanism to set state: True
blueman-applet 16.49.02 INFO     PowerManager:110 check     : callbacks done
blueman-applet 16.49.02 INFO     PowerManager:81 set_adapter_state: True
blueman-applet 16.49.02 INFO     PowerManager:174 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 16.49.05 WARNING  DBusService:42 connect_service: RecentConns plugin is unavailable
blueman-applet 16.49.05 DEBUG    Base:59 do_g_properties_changed: /org/bluez/hci0/dev_F4_60_E2_C4_96_3F {'Connected': False}
blueman-applet 16.49.05 DEBUG    NetworkManager:125 find_or_create_connection: Found existing connection with uuid 8f74a221-6eb2-4825-a034-d644ca7ab9fb
blueman-applet 16.49.05 DEBUG    NetworkManager:67 _on_device_state_changed: New: prepare Old: disconnected Reason: none
blueman-applet 16.49.06 DEBUG    NetworkManager:67 _on_device_state_changed: New: failed Old: prepare Reason: bt-failed
blueman-applet 16.49.06 DEBUG    NetworkManager:87 _on_device_state_changed: Connection failed with reason: bt-failed
blueman-applet 16.49.06 DEBUG    NetworkManager:48 _raise_or_error_handler: Connection failed with reason: bt-failed
Bus error
```

cschramm, 2020-08-19:
Doesn't look like I can reproduce that, no.

infirit, 2020-08-19:
Thx for checking. 

On a side note, I am probably going to switch away from gentoo. It is just too much work to keep it working and I don't have the time (and I am getting older and don't have the patience to watch chromium build for 10 hours). I am looking at rolling distro's but none appeal, I may try arch or debian testing.

closed 2020-08-19 by infirit

cschramm, 2020-08-19:
I guess this comes unexpected now: Go for arch. I use Arch and Debian testing / unstable in parallel and I'm absolutely sure that Arch is what you're looking for.

cschramm, 2020-08-25:
...and what always handy is https://hub.docker.com/_/python of course.

project-administrator, 2023-05-28:
Three years later I have the same issue on Debian/Kali rolling... BT networking would not connect to and Android NAP or DUN profile.

cschramm, 2023-05-29:
Most probably the `Bus error` + crash indicates a SIGBUS signal, which means that some code requested to access a completely invalid memory address. Some involved library is severely broken. You - or Kali maintainers - should be able to find it with a debugger.