blueman-project/blueman #1343: Shipped translations are outdated

ymollard, 2020-08-18:
blueman: 2.1.2
BlueZ:
Distribution: Ubuntu 20.04.1 Mate
Desktop environment: Mate

I found some untranslated French strings in blueman-applet 2.1.2 shipped with Ubuntu Mate 20.04 and hoped I could submit a translation.
In this [git repo branch 2-1-stable](https://github.com/blueman-project/blueman/tree/2-1-stable) however, these faulty strings have been translated many years ago.
After submitting a bug at Ubuntu, [they reported the shipped translations of blueman were old](https://bugs.launchpad.net/ubuntu/+source/blueman/+bug/1891950/comments/2). Any idea?

[image]

cschramm, 2020-08-18:
That conclusion seems wrong to me. Check `unmsgfmt /usr/share/locale/fr/LC_MESSAGES/blueman.mo`.

There is something wrong with the `Make Discoverable` message, though. I can even see it untranslated in the current master with `de`, also it is translated.... :thinking:

ymollard, 2020-08-18:
A few other observations from my french Ubuntu 20.04:
* [Rename device](https://github.com/blueman-project/blueman/blame/2-1-stable/po/fr.po#L143`) is untranslated
* [Help](https://github.com/blueman-project/blueman/blame/2-1-stable/po/fr.po#L80) is untranslated
* [Serial Port](https://github.com/blueman-project/blueman/blame/2-1-stable/po/fr.po#L973) is untranslated
* all "Exit" menu items are untranslated but I can't find them in the .po file
* message "Select row(s) and use <i>Control + C</i> to copy" (in the device info window from a specific device) is untranslated but I can't find them in the .po file

Also:
* `LC_ALL=C blueman-manager` does not give my an English GUI as I would expect but something mostly English with a few French inside

cschramm, 2020-08-18:
At least most of that got fixed in https://github.com/blueman-project/blueman/pull/1279. Together with #1344 and #1342 we have quite a couple of fixes incoming that we should pick for a 2.1.4, it seems. :+1:

infirit, 2020-08-18:
>LC_ALL=C blueman-manager

Don't use that, use ``LANGUAGE="en" blueman-manager``.

closed 2020-08-18 by cschramm

ymollard, 2020-08-18:
What about these ones that also appear in English?
* [Rename device](https://github.com/blueman-project/blueman/blame/2-1-stable/po/fr.po#L143`)
* [Help](https://github.com/blueman-project/blueman/blame/2-1-stable/po/fr.po#L80) 
* [Serial Port](https://github.com/blueman-project/blueman/blame/2-1-stable/po/fr.po#L973)

Also why some messages have an underscore? e.g. [this msgid](https://github.com/blueman-project/blueman/blob/cd3c0544cc2506839a726e7346fb9bfcb2de512a/po/vi.po#L745)

cschramm, 2020-08-18:
Those were fixed in #1279. I will backport it to 2.1 and see if we can get new translations in from Weblate (https://github.com/blueman-project/blueman/issues/1149#issuecomment-675757256).

"Select row(s) and use Control + C to copy" is actually another untranslated string... :confused:

cschramm, 2020-08-18:
Underscores are for keyboard access.

cschramm, 2020-08-21:
Info dialog gets fixed in #1350. Will backport that as well for a 2.1.4 as soon as I hear back from Weblate to see if there will actually be translations for those backports or not.

PR #1344 (2020-08-18): Fix an untranslated string in DiscvManager
Fixes #1343
commit 5f96f962 (2020-08-18): Fix an untranslated string in DiscvManager

Fixes #1343
commit 7db245b3 (2020-08-20): Fix an untranslated string in DiscvManager

Fixes #1343