blueman-project/blueman #1348: Broadcom Corp. BCM2046B1 Not Working

JordanPlayz158, 2020-08-20:
blueman: 2.1.3-1
BlueZ: 5.54-2
Distribution: Manjaro
Desktop environment: XFCE

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

(I unplugged and replugged in my adapter and tried to scan for devices
blueman-applet --loglevel debug:
https://pastebin.com/yspGvPxs

bluetoothctl list (didn't think 1 line of text warranted a pastebin): 
Controller 98:00:20:04:00:00 Jordan-PC [default]

(I unplugged, replugged, and scanned)
Output in "journalctl -f" (Think this contains most useful information):
https://pastebin.com/qnT4F4Uv

I had to load up the device in a windows vm because the seller didn't know what Broadcom chip it uses (I tried lsusb and it only said Broadcom Corp.), found out it uses the BCM2046B1

cschramm, 2020-08-20:
First and foremost, this is clearly not a blueman issue, rather most probably firmware related. There are all sorts of issues with Broadcom Bluetooth chips and BCM2046 seems to be one of the bestest of them. You can even find a handful of issues that refer to it right here in blueman's issue tracker.

Regarding the blueman-applet log I see a rather usual startup, 10 seconds later an adapter shows up and another 24 seconds later it disappears again. There's nothing about an attempted discovery. :thinking:

Regarding the syslog, I fully agree that `kernel: Bluetooth: hci0: command 0x0401 tx timeout` is concerning, yes.

Unfortunately I have no idea how and what firmware Arch ships for those chips but I think your only chance will be to find a replacement.

JordanPlayz158, 2020-08-20:
Well the "kernel: Bluetooth: hci0: command 0x0401 tx timeout" happens when I try to scan via the gui (blueman-manager) so I think if we could fix that shouldn't the bluetooth work?