blueman-project/blueman #1349: Drop Python 3.6 support?

cschramm, 2020-08-20:
https://github.com/blueman-project/blueman/pull/1341 tries to make `PluginManager` generic (again) and as it's also a `GObject` and `Generic` is of type `GenericMeta` in Python 3.6, it fails in Python 3.6 without a dedicated meta class as proposed in https://github.com/blueman-project/blueman/issues/1130#issuecomment-548748082. However, that solution does not work for me either as Python 3.6 gives me that:
```
Traceback (most recent call last):
  File "blueman/main/PluginManager.py", line 35, in <module>
    class PluginManager(GObject.GObject, Generic[_T], metaclass=PluginManagerMeta):
  File "/usr/local/lib/python3.6/typing.py", line 978, in __new__
    self = super().__new__(cls, name, bases, namespace, _root=True)
  File "/usr/local/lib/python3.6/typing.py", line 137, in __new__
    return super().__new__(cls, name, bases, namespace)
  File "/usr/local/lib/python3.6/abc.py", line 133, in __new__
    cls = super().__new__(mcls, name, bases, namespace, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/gi/types.py", line 253, in mro
    return mro(cls)
  File "/usr/local/lib/python3.6/site-packages/gi/types.py", line 311, in mro
    not_head = [s for s in bases_of_subclasses if candidate in s[1:]]
  File "/usr/local/lib/python3.6/site-packages/gi/types.py", line 311, in <listcomp>
    not_head = [s for s in bases_of_subclasses if candidate in s[1:]]
  File "/usr/local/lib/python3.6/typing.py", line 1092, in __eq__
    if self.__origin__ is None or other.__origin__ is None:
AttributeError: type object 'PluginManager' has no attribute '__origin__'
```
I might be doing something really stupid there, not sure. Anyway, as this blocks me a little bit, I was thinking about just dropping support for Python 3.6. I checked the "rolling repositories" that have blueman 2.1.3 according to https://repology.org/project/blueman/versions and all of them currently have Python 3.8 or are even migrating to 3.9 already. So basically the only group targeted by support for Python 3.6 and 3.7 is people installing blueman from source using those versions for whatever reason and that sounds reeaally limited.

cschramm, 2020-08-24:
So... I found a solution that seems to be working in https://github.com/blueman-project/blueman/blob/d36b7cdb0eca468a0e02e0cc2e3713ed798c5bc1/blueman/main/PluginManager.py#L26. I have no idea what side effects that hack might have, though... Also, `from typing import GenericMeta  # type: ignore` will need some thoughts as mypy reports an unused ignore here when run in Python 3.6.

infirit, 2020-08-24:
tbh, all of this goes way over my head. So if this works and it does not break the working of blueman 👍

cschramm, 2020-08-26:
Not saying that we should not consider dropping support for Python 3.6 anymore but I found a way around the generic in #1341, so my major pain point with it is gone. :slightly_smiling_face:

closed 2020-08-26 by cschramm