blueman-project/blueman #1362: Applet Interface language

bentzys, 2020-08-24:
blueman: 2.1.2
BlueZ:
Distribution: 
Desktop environment: kubuntu 20.04

* [ *] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

How can I force English as Interface language despite OS locale?

cschramm, 2020-08-24:
Just set the desired locale as usual, e.g. `LANG=en_US.UTF-8 blueman-applet`.

closed 2020-08-24 by cschramm

bentzys, 2020-08-24:
Sorry, not clear.
I installed blueman via apt.
My applet runs at KDE's system tray.
Where the "blueman-applet" call is so I can prepend "LANG=en_US.UTF-8" ?
Thanks in advance!
--bentzy

cschramm, 2020-08-24:
blueman-applet most probably gets started automatically by KDE picking up `/etc/xdg/autostart/blueman.desktop`. KDE has an autostart manager, see [docs.kde.org](https://docs.kde.org/stable5/en/kde-workspace/kcontrol/autostart/index.html). Unless you can configure environment variables for specific entries there (I guess not), you can disable the standard blueman entry and add your own that uses `env` to do the job, so that its command line looks like `env LANG=en_US.UTF-8 blueman-applet` (or change `/etc/xdg/autostart/blueman.desktop` accordingly but then you're getting into a fight with your package management).

To be honest, making different applications within a single desktop session run with different languages sounds like a really strange use case to me from a user perspective. May I ask what you're actually trying to achieve?

bentzys, 2020-08-25:
Issue solved:
Findings:
Culprit was the "LANGUAGE" env variable. It is surprisingly not set by any text file (.bash_rc, /etc/profile, /etc/profile.d/* or others) , but it is set by KDE according your selections in KDE System settings- > Configure Plasma Translations menu. Since I need hebrew in my workstation, "echo $LANGUAGE" returns "en_US:he" (first is default). But no matter which is the default LANGUAGE, blueman-applet showed hebrew!
Solution: at /etc/xdg/autostart/blueman.desktop
Exec= LANGUAGE=en blueman-applet
I do not know if the issue is in blueman-applet or at KDE system-tray, but no other system tray apple showed hebrew...
Anyway, may be helpful to have a way to set UI language as a preference setting
--bentzy

cschramm, 2020-08-25:
Now we're talking. :smile: 
You just found a nice little edge case in our localization.
While `LANGUAGE=en_GB:he` will primarily use the `en_GB` locale that we ship, `LANGUAGE=en_US:he` or `LANGUAGE=en:he` will just fall back to `he` as we ship neither of those `en` locales. `en` is just our default language, included in our code, not shipped as a `blueman.mo` file and it seems to me that `gettext` just has no way of knowing that it is and thus thinks "`en` / `en_US` is not available but we can use `he`". I'm not sure how to tell it, right now. An obvious solution would be to ship a redundant `en` locale as well. :thinking:

cschramm, 2020-08-25:
It looks like this is a common scheme, e.g. https://unix.stackexchange.com/questions/181857/the-language-locale-variable-how-to-set-english-as-primary-language-a-bug-i/181867#181867 discusses the same topic and advises to use `LANGUAGE=en_US:C:he`, assuming that C, the base language, is English. (By the way, I would just use `en` instead of `en_US` to have a better chance of finding something). I wonder in what cases `he` will actually come into play with that setting, though, as C seems to always terminate the lookup...

bentzys, 2020-08-25:
But...I do not set "LANGUAGE"...It is KDE...
I can add a lot of languages in any order. But not "C"  ;-)

cschramm, 2020-08-25:
Well, actually if you assume that the base language, denoted as `C` is `en` and gettext at least used to make that assumption, and you prefer `en_US` over `en` / `C` over `he`, then `LANGUAGE` can just be set to `en_US` as you'll get the `C` fallback anyway and `he` behind `C` will not have an effect at least on gettext.

bentzys, 2020-08-25:
Can't choose "en' nor 'C'
English options at KDE are American English (en_US) and British English (en_GB).
I solved the issue with "LANGUAGE=en blueman-applet"
And in the meantime found a bug in glibc (https://bugs.launchpad.net/ubuntu/+source/glibc/+bug/1892825) ;-)
Thanks for blueman!

bentzys, 2020-08-26:
BTW, a minor issue:
at applet tooltip, I see the html tags, as following:
[image]

cschramm, 2020-09-03:
I guess you're using our `AppIndicator` plugin. We use libappindicator there and that is one piece of badly designed interface tightly coupled to indicator-application that it was initially build for. There is no differentiation between plain text and markup, just as single title, and tags in it get weird treatment in application-indicator as it strips some and shows others, see e.g. https://bugs.launchpad.net/ubuntu/+source/indicator-application/+bug/522149. Some implementations copy that behavior, some even support markup in the title, some expect plain text like yours (it's Plasma 5, right?).

When I just had another look it at, I realized that it's probably plain stupid that we're still using libappindicator, though, as there is an underlying DBus protocol with a much better interface than libappindicator's and the freedesktop.org StatusNotifierItem specification even shows the common understanding (although in reality things are `org.kde`, not `org.freedesktop` :smile:) and that specification does have a tooltip property with explicit and well-defined markup support besides the title property that it just describes as "a name that describes the application". I guess we should just use the DBus protocol directly to be able to use the tooltip property and get better results.

I just created a small Python script to test it: https://gist.github.com/cschramm/c6cc7cea909d7a40e6f5a976dbaf484f. Just run and check what tooltip and title stuff it gives you for the blueman icon that it adds.

bentzys, 2020-09-03:
Indeed, plasmashell 5.18.5
I've ran the script and get the icon at tray...but no title/tooltip.
Clicking (using all three buttons) at icon I get ,respectively , the following output:
```
bentzy@bentzy-nb:~$ python3 snitest.py
(<Gio.DBusConnection object at 0x7fbf58f3c200 (GDBusConnection at 0x26ae070)>, ':1.22', '/org/blueman/sni', 'org.kde.StatusNotifierItem', 'Activate', GLib.Variant('(ii)', (59, 869)), <Gio.DBusMethodInvocation object at 0x7fbf57f8df00 (GDBusMethodInvocation at 0x7fbf48014330)>)
(<Gio.DBusConnection object at 0x7fbf58f3c200 (GDBusConnection at 0x26ae070)>, ':1.22', '/org/blueman/sni', 'org.kde.StatusNotifierItem', 'ContextMenu', GLib.Variant('(ii)', (59, 869)), <Gio.DBusMethodInvocation object at 0x7fbf57f46fc0 (GDBusMethodInvocation at 0x7fbf48014830)>)
(<Gio.DBusConnection object at 0x7fbf58f3c200 (GDBusConnection at 0x26ae070)>, ':1.22', '/org/blueman/sni', 'org.kde.StatusNotifierItem', 'SecondaryActivate', GLib.Variant('(ii)', (59, 869)), <Gio.DBusMethodInvocation object at 0x7fbf57f45e40 (GDBusMethodInvocation at 0x7fbf480148a0)>)

```

infirit, 2020-09-03:
The biggest advantage of libapindicator is that it providea a fallback to GtkStatusIcon when indicator is not available.

I am not sure if sni has this but if it does let's move to it asap.

micw, 2021-01-07:
IMO has libapindicator more disadvantages than advantages. I recently installed network-manager-applet which pulled libapindicator as dependency. Since that, I have indicator issues with many applications (including blueman-applet). One thing is that left click stopped to work, both mouse buttons trigger the popup menu. Another is that the menu shows up directly under the mouse pointer. If I release the mouse, the action (in this case "close application") is triggered. So on a sinlge mouse click on blueman-applet with libapindicator, the application is closed now.

I uninstalled libapindicator and all works pretty well now.