blueman-project/blueman #1363: Contextualize and internationalize errors

ymollard, 2020-08-24:
Following https://github.com/blueman-project/blueman/pull/1347#issuecomment-679111971, here's a discussion about error messages such as in the screenshot below.

The point is to figure out how we could make errors more understandable both in terms of content (Contextualize kernel errors) and form (translate them into the language of the UI), with a minimum effort and maintenance need.

[image]

Here's a list of common errors reported in [~800 Github issues](https://github.com/blueman-project/blueman/issues):
* `Protocol not available`
* `No such file or directory`
* `Input/output error`
* `Host is down`
* `Resource temporarily unavailable` (e.g. incorrect/cancelled pairing key)
* `Software caused connection abort` (e.g. Bluetooth turned off during connection)

cschramm, 2020-08-25:
Those are
* ENOPROTOOPT
* ENOENT
* EIO
* EHOSTDOWN
* EAGAIN
* ECONNABORTED

I fully agree that the standard `strerror` messages are rather misleading in these contexts.

As discussed in #1347 we are left with matching the error messages as we do not have the error codes, so I guess somebody should at least check musl if it uses the same descriptions for those codes.

We also need to gather the contexts that these errors show up in to find proper descriptions. Are they all from that same "Connection Failed" context?

ymollard, 2020-08-31:
> We also need to gather the contexts that these errors show up in to find proper descriptions. Are they all from that same "Connection Failed" context?

It's hard to say, I only grep-ed a bunch of issues searching for dbus errors. How can we figure out?

I've just edited the initial message by adding `Resource not ready` if you wanna extend the list of errnos in your message ; I got this one today ([context here](https://github.com/blueman-project/blueman/issues/1368))

cschramm, 2020-11-21:
I dug into this in the context of generic device connection (opposed to network or serial service connection, discovery, cancellation etc.) a bit and collected the following about those messages so far:

`Protocol not available` is the message for ENOPROTOOPT in glibc, uClibc, and musl and means that no audio endpoints are registered to BlueZ (which is typically done by pulseaudio's Bluetooth module).

`Input/output error` is the message for EIO in glibc and uClibc. The musl message for it is `I/O error`. It means all sorts of low level stuff going wrong and oftentimes hints at driver / firmware issues.

`Host is down` is the message for EHOSTDOWN in glibc, uClibc, and musl and represents Bluetooth errors 0x04 and 0x3c (advertising timeout)

`Software caused connection abort` is the message for ECONNABORTED in glibc and uClibc. The musl message for it is `Connection aborted` and represents Bluetooth error 0x16 (connection terminated by local host)

`Resource not ready` comes directly from bluetoothd (as `org.bluez.Error.NotReady`, not `org.bluez.Error.Failed`, https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/error.c?id=35a2c50437cca4d26ac6537ce3a964bb509c9b62#n108). I'm not sure if it is really expected or. There is one reference in #443 related to a bug that's fixed and another one without information in #824 but I'm rather sure it was a 2.0 version and thus lacked the same fix.

I cannot find anything like "org.bluez.Error.Failed: No such file or directory" or "org.bluez.Error.Failed: Resource temporarily unavailable" in our GitHub issues. Not sure why those messages are on the list.

ymollard, 2020-11-25:
> I cannot find anything like "org.bluez.Error.Failed: No such file or directory" or "org.bluez.Error.Failed: Resource temporarily unavailable" in our GitHub issues.

`Resource temporarily unavailable` is triggered for instance when the pairing key is incorrect or the user cancelled pairing.

As for `No such file or directory`, I probably went through this one during my testing session but I didn't note more context. I'll post updates here if I can reproduce it again.

cschramm, 2020-11-25:
> `Resource temporarily unavailable` is triggered for instance when the pairing key is incorrect or the user cancelled pairing.

Where? How?

I can found three cases with at least some context in our issues:

#882 -> firmware-ish issues
#1001 -> supposedly solved by reinstalling pulseaudio :shrug: 
Some mentions in #650 where it seems to actually be a response to a connection attempt and even a proper log in https://github.com/blueman-project/blueman/issues/650#issuecomment-280505939. :raised_hands: I'll look into what triggers it (if it still exists in recent BlueZ versions.)

cschramm, 2020-11-26:
OK, got it. `Resource temporarily unavailable` (EAGAIN) shows up on unexpected errors when setting up an audio connection. If the device is not yet paired, the audio connection gets such an unexpected error (ECONNRESET) if pairing is rejected on any side.

This one is rather tricky as in that specific case (cancelled pairing while connecting to an audio device) we can just suppress the error completely as we do with `Cancelled` but looking at #650 there are other occurrences of that error as well that are basically just like EIO, so we cannot suppress it in general.

PR #1444 (2020-11-26): Translate EAGAIN
A typical occurrence is when pairing gets cancelled while connecting to an unpaired audio device but there are other occurrences related to BlueZ / driver / firmware issues (see #650) and it can always hint at generic unexpected (audio) connection errors similar to EIO.

Closes #1363
commit ed8d903f (2020-11-26): Translate EAGAIN

A typical occurrence is when pairing gets cancelled while connecting to an unpaired audio device but there are other occurrences related to BlueZ / driver / firmware issues (see #650) and it can always hint at generic unexpected (audio) connection errors similar to EIO.

Closes #1363
closed 2020-11-26 by cschramm