blueman-project/blueman #1368: urb ... failed to resubmit (113) + UnicodeDecodeError

ymollard, 2020-08-31:
For no reason I'm suddenly getting errors from the kernel when bluetooth service is (re)started. Journalctl says:
```
sept. 01 01:21:41 laptop systemd[1]: Started Hostname Service.
sept. 01 01:21:42 laptop dbus-daemon[966]: [system] Activating via systemd: service name='org.blueman.Mechanism' unit='blueman-mechanism.service' requested by ':1.2539' (uid=1000 pid=352047 comm="/usr/bin/python3 /usr/bin/blueman-applet " label="unconfined")
sept. 01 01:21:42 laptop systemd[1]: Starting Bluetooth management mechanism...
sept. 01 01:21:42 laptop blueman-mechanism[366359]: Unable to init server: Could not connect: Connection refused
sept. 01 01:21:42 laptop blueman-mechanism[366359]: Unable to init server: Impossible de se connecter : Connection refused
sept. 01 01:21:42 laptop blueman-mechani[366359]: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
sept. 01 01:21:42 laptop dbus-daemon[966]: [system] Successfully activated service 'org.blueman.Mechanism'
sept. 01 01:21:42 laptop systemd[1]: Started Bluetooth management mechanism.

sept. 01 01:21:43 laptop bluetoothd[366339]: Loading LTKs timed out for hci0
sept. 01 01:21:44 laptop kernel: Bluetooth: hci0: urb 00000000153bcec7 failed to resubmit (113)

```
In the GUI, any connection tentative later on then results in `Resource Not Ready`. 

Although any idea is welcome about this, my issue also concerns how blueman 2.1.2-1 reacts to this strange bug from the kernel. 
Indeed, in generates such Python errors `UnicodeDecodeError: 'utf8' codec can't decode byte 0xa5 in position 0: invalid start byte`. I assume that try/catching decode instructions would be the right fix, at least to prevent Apport from generating crash reports.  

If someone can consult the Apport reports gatherd by Ubuntu 20.04, I've sent of few of them in the last 12hrs.

cschramm, 2020-09-01:
See https://github.com/blueman-project/blueman/issues/1241#issuecomment-655923852 for the blueman part.

"failed to resubmit" gives me a lot of good matches on Google, not sure if they include a useful solution beside replacing the hardware, though. :smile:

closed 2020-09-01 by cschramm

ymollard, 2020-09-01:
A reboot fixed the `failed to resubmit` issue, but what about the `UnicodeDecodeError`, shouldn't it be try/catched?

cschramm, 2020-09-02:
See the linked comment. Our code is not involved and there is probably not even a way to catch the exception for us except of a generic `sys.excepthook`. Apart from that, it does not do harm, and I would not really see stuff like Apport annoying the user as the problem of other applications than Apport itself. Ubuntu can add `Conflicts: apport` to their blueman package to avoid it. 🤭 Even if one sees it as others' problem, then probably PyGObject should catch that bug, not us.