blueman-project/blueman #1370: two BTZ icons in tray when booting with adapter inserted

michaelhmich, 2020-09-02:
blueman: 2.12
BlueZ: bluez (5.53-0ubuntu3) focal;
Distribution: LM20 kernel 5.8.5
Desktop: Kde Plasma 5.18,5

when i boot with the bt adapter inserted I get 2 btz icons on the desktop system tray, both work identical and have the same dd menu

when I insert it after log into kde it is only one

cschramm, 2020-09-02:
Two blueman icons should probably be fixed by #1304 but it sounds really strange that it seems to depend on the presence of an adapter. I assume nothing shows up in any case if you disable blueman in KDE's autostart manager, right?

closed 2020-09-17 by github-actions[bot]

michaelhmich, 2020-10-12:
sorry for the late replay, googe created without advice a new forum folder
and your comment was there, and yes, when BTZ is disabled no icon shows up,
I can not tell if it is solved because since I uninstalled KDEConnect my
problem went with it away, as you said #1305 sounds to be the same problem
just triggered in another way

thank you