blueman-project/blueman #1373: 2.1.4

cschramm, 2020-09-05:
Some fixes piled up in https://github.com/blueman-project/blueman/compare/2.1.3...2-1-stable so that I think it's time for a 2.1.4.

Some stuff that bugs me:
* @infirit: It seems like https://github.com/blueman-project/blueman/commit/778b2ad94265dda6f170d68b6b629dfa68225e3b ended up in the wrong section. Looking at the commits those things were *not* part of 2.1.3 but will be in 2.1.4, right?
* We backported some "make strings translatable" improvements and Weblate enabled me to set up https://hosted.weblate.org/projects/blueman/stable/. Obviously untranslated stuff did not get translated a lot yet, but what bugs me in particular is that there are some untranslated strings in the stable component that have translations in the master component and it would make total sense to just copy them all for all languages but it seem like you can only do that one by one. (@ymollard: To fix https://github.com/blueman-project/blueman/issues/1343 for fr, would you mind going through https://hosted.weblate.org/translate/blueman/stable/fr/?q=check:inconsistent? You will probably want to just apply the master translations at least for stuff that's untranslated in stable yet) (Translations coming in in #1366)

ymollard, 2020-09-05:
I did my best for french strings.

P.S.: When the translated string is the same as the English string (e.g. "OBEX"), is it better to leave the field empty or copy/paste it?

Also:

> Connects auto connect profiles A2DP source, A2DP sink, and HID

Aren't that too much "connect" words in the same sentence? Does it mean "Enable autoconnect"?

cschramm, 2020-09-06:
That's the tooltip for the the connect item in blueman-manager's device menu, so it describes what that item does and it's not about enabling anything but actually connecting profiles.

infirit, 2020-09-08:
I hope to spend some time on this coming weekend.

cschramm, 2020-09-12:
> there are some untranslated strings in the stable component that have translations in the master component and it would make total sense to just copy them all for all languages

I've just done that by hand. 🙂 (#1366 lacks squashing for unknown reasons.)

fitojb, 2020-09-16:
Hi everyone! Pls let me this weekend to finish up the Spanish translation. (I’m Adolfo, from Launchpad and later Transifex ;)

closed 2020-09-27 by cschramm