blueman-project/blueman #1374: Feature Request: desktop notifications via notify-send

hissssst, 2020-09-05:
First of all, thank you for this awesome software

I'd like to open discussion about desktop notifications with blueman.
It would be nice to have notifications sent when device connects/disconnects

infirit, 2020-09-08:
I am not against adding a notification but not without a proper use case. Can you be more specific in which situation this may be useful and how it would help you?

hissssst, 2020-09-09:
I would like to receive notifications when device connects/disconnects. It's useful because I am connected to multiple bt-headsets at one time and some of them can disconnect when they are out of battery.

It would be nice to receive notification like `Device Headset Pro (12:34:56:78:ab:cd) disconnected`

cschramm, 2020-09-12:
Makes perfect sense to me. Usually you do not need it as the status icon change is sufficient if you use just one device at a time but you lack that feedback if you use multiple.

PR #1424 (2020-11-08): Add plugin for connection notifications
Closes #1374

Causes some fuss in combination with #1418 and #1414. We should probably find some way to merge those up to three notifications.
commit 3db008b7 (2020-11-08): Add plugin for connection notifications

Closes #1374

Causes some fuss in combination with #1418 and #1414. We should probably find some way to merge those up to three notifications.
closed 2020-11-08 by cschramm