blueman-project/blueman #1378: Bluetooth 5 (Asus BT-500)

infirit, 2020-09-07:
My usb Bluetooth 4.0 dongle started acting up and I decided to buy a new one. I came across the Asus BT-500 which is a Bluetooth 5.0 dongle with a gpl kernel driver. It is in pre-order/coming in September. Bluetooth 5 is not really changing a lot and I have no devices that support it. But if you want me to provide some details or test some thing let me know.

cschramm, 2020-09-27:
:man_shrugging: Does it work at all? :smile:

infirit, 2020-09-28:
I am still waiting on the damn thing. Expected to ship in October so I got
cheap csr dongle so I have working Bluetooth 4 again.

On a side note, the csr dongle does work much better than the brcm dongle.

infirit, 2020-11-04:
Finally got it in today, verbose lsusb output https://gist.github.com/infirit/231280f03ea2a936be79e0a8cbc07585

ps, I'll make some time to go over all the PR and issue comment this weekend. Sorry for the delay in review.

OJFord, 2020-11-12:
I got a 5.1 card (Gigabyte GC-WB1733D-I) today, it works, but I'm having trouble verifying that my headset that claims to be 5 really is (the card is, from `hciconfig`/`btmgmt`) and if it is that the connection is making any use of it.

It's an upgrade from a 2.0 for me, so I was expecting/hoping that the HSP/HFP headset mode would be improved, in terms of audio quality, but it seems the same.

Blueman reports ~16.6KB/s up and down, and  ~41KB/s up, 370B/s down in A2DP mode. Range does seem improved, but that could just be the proper antenna it has now, vs. the tiny PCB-inside-USB-port thing I was using before. The rates mentioned are from sitting right next to it. Theoretically we'd expect more like 125KB/s each way in headset & 250KB/s A2DP mode I believe.

Having said that, I'm not even sure the modes are the same. Perhaps the mere mention of HSP/HFP/A2DP was enough for folks more familiar to know something's not right here!

Appreciate any pointers and happy to test or provide whatever I can. Cheers,

cschramm, 2020-11-13:
@OJFord: `btmon` should show some information when connecting, at least the PHYs. I think LE Coded should also show up there if used but I'm not sure where exactly.

OJFord, 2020-11-13:
Thanks @cschramm, this maybe isn't the place for it, I don't want to hijack the discussion for my own personal troubleshooting, but this is `btmon` during connection if it means anything to you:

<details>
<pre>
Bluetooth monitor ver 5.55
= Note: Linux version 5.9.8-arch1-1 (x86_64)                                                                                                                                 0.217391
= Note: Bluetooth subsystem version 2.22                                                                                                                                     0.217392
= New Index: BC:54:2F:62:9E:30 (Primary,USB,hci0)                                                                                                                     [hci0] 0.217393
= Open Index: BC:54:2F:62:9E:30                                                                                                                                       [hci0] 0.217393
= Index Info: BC:54:2F:62:9E:30 (Intel Corp.)                                                                                                                         [hci0] 0.217393
@ MGMT Open: bluetoothd (privileged) version 1.18                                                                                                                   {0x0001} 0.217394
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                     #1 [hci0] 0.300787
        Status: Success (0x00)
        Handle: 256
        Reason: Connection Terminated By Local Host (0x16)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                            {0x0001} [hci0] 0.300824
        BR/EDR Address: 00:25:BB:06:6D:62 (INNERINT Co., Ltd.)
        Reason: Connection terminated by local host (0x02)
> HCI Event: Connect Request (0x04) plen 10                                                                                                                       #2 [hci0] 10.147694
        Address: 00:25:BB:06:6D:62 (INNERINT Co., Ltd.)
        Class: 0x240418
          Major class: Audio/Video (headset, speaker, stereo, video, vcr)
          Minor class: Headphones
          Rendering (Printing, Speaker)
          Audio (Speaker, Microphone, Headset)
        Link type: ACL (0x01)
< HCI Command: Accept Connection Request (0x01|0x0009) plen 7                                                                                                     #3 [hci0] 10.147751
        Address: 00:25:BB:06:6D:62 (INNERINT Co., Ltd.)
        Role: Master (0x00)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                         #4 [hci0] 10.148688
      Accept Connection Request (0x01|0x0009) ncmd 1
        Status: Success (0x00)
> HCI Event: Role Change (0x12) plen 8                                                                                                                            #5 [hci0] 10.283728
        Status: Success (0x00)
        Address: 00:25:BB:06:6D:62 (INNERINT Co., Ltd.)
        Role: Master (0x00)
> HCI Event: Vendor (0xff) plen 4                                                                                                                                 #6 [hci0] 10.460692
      Intel PTT Switch Notification (0x26)
        Handle: 256
        Packet type table: Enhanced data rate (0x01)
> HCI Event: Connect Complete (0x03) plen 11                                                                                                                      #7 [hci0] 10.463691
        Status: Success (0x00)
        Handle: 256
        Address: 00:25:BB:06:6D:62 (INNERINT Co., Ltd.)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
< HCI Command: Read Remote Supported Features (0x01|0x001b) plen 2                                                                                                #8 [hci0] 10.463892
        Handle: 256
> HCI Event: Command Status (0x0f) plen 4                                                                                                                         #9 [hci0] 10.464684
      Read Remote Supported Features (0x01|0x001b) ncmd 1
        Status: Success (0x00)
> HCI Event: Max Slots Change (0x1b) plen 3                                                                                                                      #10 [hci0] 10.469686
        Handle: 256
        Max slots: 5
> HCI Event: Read Remote Supported Features (0x0b) plen 11                                                                                                       #11 [hci0] 10.470715
        Status: Success (0x00)
        Handle: 256
        Features: 0xff 0xfe 0x8f 0xfe 0xdb 0xff 0x5b 0x87
          3 slot packets
          5 slot packets
          Encryption
          Slot offset
          Timing accuracy
          Role switch
          Hold mode
          Sniff mode
          Power control requests
          Channel quality driven data rate (CQDDR)
          SCO link
          HV2 packets
[... 118 lines cut ...]
        Status: Success (0x00)
        Address: 00:25:BB:06:6D:62 (INNERINT Co., Ltd.)
        Name: TAOTRONICS SoundElite 72
[... cut ...]
< ACL Data TX: Handle 256 flags 0x00 dlen 16                                                                                                                     #30 [hci0] 10.499063
      L2CAP: Connection Response (0x03) ident 1 len 8
        Destination CID: 64
        Source CID: 833
        Result: Connection successful (0x0000)
        Status: No further information available (0x0000)
</pre>
</details>

The 'LE' and 'enhanced data rate 2/3 Mbps' sound like the headset is indeed 5 to me? So I'm not sure if it's my expectation that's at fault or the data rate I actually see when using for audio really is lower than it should/could be.

cschramm, 2020-11-13:
LE is 4.0+, EDR ist 2.0+. What would be interesting are the 5.0+ `LE 2M PHY` (extended bitrate) and `LE Coded PHY` features. We should see them on a `LE Read Remote Used Features` event but the command is not issued. Maybe you need to turn the remote device off (and remove it locally) and back on to trigger it.

`bluetoothctl show` should show them as `SupportedSecondaryChannels`. I'm far from an expert but to me your monitoring output looks like LE isn't used at all although it seems supported. I'd expect to see an `LE Advertising Report` / `LE Extended Advertising Report` event and an `LE Create Connection` / `LE Extended Create Connection` command.

OJFord, 2020-11-13:
Thanks, I see `LE 1M` (*not* `2M` - I don't know what this means/if significant) for `Primary PHY`, and `Create Connection` is not `LE Create Connection`, which I suppose means 'LE isn't used at all although it seems supported', as you said?

<details>
<pre>
> HCI Event: LE Meta Event (0x3e) plen 56                   #18 [hci0] 7.799252
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0013
            Props: 0x0013
              Connectable
              Scannable
              Use legacy advertising PDUs
            Data status: [0;32mComplete[0m
          Legacy PDU Type: ADV_IND (0x0013)
          Address type: Public (0x00)
          Address: 00:25:BB:06:6D:62 (INNERINT Co., Ltd.)
          Primary PHY: LE 1M
          Secondary PHY: No packets
          SID: no ADI field (0xff)
          TX power: 127 dBm
          RSSI: -50 dBm (0xce)
          Periodic advertising invteral: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x1e
        03 03 c1 ab 19 09 54 41 4f 54 52 4f 4e 49 43 53  ......TAOTRONICS
        20 53 6f 75 6e 64 45 6c 69 74 65 20 37 32         SoundElite 72
[...]
< HCI Command: Create Connection (0x01|0x0005) plen 13     #34 [hci0] 13.330873
        Address: 00:25:BB:06:6D:62 (INNERINT Co., Ltd.)
        Packet type: 0xcc18
          DM1 may be used
          DH1 may be used
          DM3 may be used
          DH3 may be used
          DM5 may be used
          DH5 may be used
        Page scan repetition mode: R2 (0x02)
        Page scan mode: Mandatory (0x00)
        Clock offset: 0x0000
        Role switch: Allow slave (0x01)
</pre></details>

cschramm, 2020-11-13:
Exactly, that's how I would read it. Fun fact: The non-LE connection is initiated by the remote device although it would support LE. I have no clue why it decides to do that. :shrug:

OJFord, 2020-11-13:
Thanks again. I wonder if it's `bluez`'s issue or the headset's, maybe I can test with a Mac (which is probably not v5 but 4 at least).

Does `LE 1M` as opposed to `LE 2M` indicate that it's v4 rather than v5? I wasn't sure of the significance of that.

cschramm, 2020-11-13:
Oh, I did not realize that there's new data in your latest output. There's an undirected / broadcast `LE Extended Advertising Report` from the remote device. That is a 5.0 feature but it advertises only LE 1M, neither LE 2M for extended bitrate nor LE Coded for extended range, so it does not seem to make use of the more interesting 5.0 features. Your system then issues `Create Connection` which is a non-LE command. In this case it seems more like your system is at fault but I don't know what happened in between.

infirit, 2021-03-04:
Ok, finally got it working. Below bluetoothctl info dump.

```
	Name: localhost
	Alias: bt500
	Class: 0x003e0104
	Powered: yes
	Discoverable: yes
	DiscoverableTimeout: 0x00000000
	Pairable: yes
	UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
	UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
	UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
	UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: NAP                       (00001116-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0538
	Discovering: no
	Roles: central
	Roles: peripheral
Advertising Features:
	ActiveInstances: 0x00 (0)
	SupportedInstances: 0x04 (4)
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name
	SupportedSecondaryChannels: 1M
	SupportedSecondaryChannels: 2M
	SupportedSecondaryChannels: Coded
```

nagynorbert97, 2021-04-01:
Hi,

I just have a question and i hope i didn't miss the answer below. I would need this dongle to use the xbox one controller with my pc, i would like to know if this supports multiple devices at the same time. It is kinda 5.0 feature. What you guys think, can i use 2 controllers at the same time with this?

cschramm, 2021-04-01:
@nagynorbert97: I'm not aware of limitations with multiple connected devices in general. What 5.0 feature do you mean?

nagynorbert97, 2021-04-01:
> @nagynorbert97: I'm not aware of limitations with multiple connected devices in general. What 5.0 feature do you mean?

Oh wow thanks for the quick reply. Sometimes "simultaneous multi device support" or something like that is listed in the item description of BT 5.0 donglesbut i havent found anything like that.

cschramm, 2021-04-01:
@nagynorbert97: I don't see why two controllers at the the same time should not work in general, independent of the Bluetooth version. :shrug:

infirit, 2021-04-02:
Don't take what manufacturers put on the box too seriously. I recently saw a speaker lamp that said it was bluetooth 5.0. I was curious and it was cheap (hint) so I bought it. There was nothing Bluetooth 5.0 specific at all, just meant it worked on Bluetooth 5.0.

Anyway, I wouldn't recommend this dongle as support in the Linux kernel is not there yet.

As for multiple devices, all major consoles use Bluetooth to connect wireless controllers. It has been like this at least since the ps3/wii/xbox times.

closed 2021-06-09 by github-actions[bot]

joakim-tjernlund, 2021-06-26:
> Don't take what manufacturers put on the box too seriously. I recently saw a speaker lamp that said it was bluetooth 5.0. I was curious and it was cheap (hint) so I bought it. There was nothing [...]

I sent in patches/firmware for BT500 a few weeks ago, it is  in bluetooth-next and linux-firmware.

elupus, 2021-07-19:
> 
> I sent in patches/firmware for BT500 a few weeks ago, it is in bluetooth-next and linux-firmware.

@joakim-tjernlund
May i ask why it was added to the blacklist table for btusb? Seem strange, but I'm not versed in how these kernel driver work.

joakim-tjernlund, 2021-10-05:
@elupus , blacklist? You may refer to to the old way when one needed an out of tree linux driver and had to blacklist the
in tree BT usb stack.