blueman-project/blueman #1379: Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)

lucassmacedo, 2020-09-08:
Hi! I have a Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode) on Ubuntu 20.04, it's recognized but the bluetooth isn't turn on. 
➜  ~ sudo hciconfig hci0 up
Can't init device hci0: Operation not supported (95)

--

➜  ~ hcitool dev
Devices:

--

--➜  ~ sudo rfkill list  
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no

--

➜  ~ blueman-manager          
blueman-manager version 2.1.2 starting
blueman-manager 09.07.03 ERROR    Manager:118 on_dbus_name_appeared: Default adapter not found, trying first available.
blueman-manager 09.07.03 ERROR    Manager:122 on_dbus_name_appeared: No adapter(s) found, exiting

Please i need help.

infirit, 2020-09-08:
Distro? Blueman version....?

Are you sure it is in hci mode? To be sure run ``hid2hci -0``. If that does not change see troubleshooting on the GitHub wiki.

lucassmacedo, 2020-09-08:
Yes,  HCI Mode.

➜  ~ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0000:3825   USB OPTICAL MOUSE
Bus 001 Device 003: ID 1c4f:0002 SiGma Micro Keyboard TRACER Gamma Ivory
Bus 001 Device 002: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

infirit, 2020-09-08:
These type of problems are best addressed on the ubuntu forums. Blueman is just a frontend to BlueZ. 

I did have a quick look and the problem may be related to you having a fake csr dongle. See https://askubuntu.com/questions/1168123/how-do-i-get-my-bluetooth-device-working

Blueman can not work until you can get the hci up. Once you get that fixed I suspect blueman will work without problem.

closed 2020-09-08 by lucassmacedo