blueman-project/blueman #1380: Missing option to connect headset

tracek, 2020-09-09:
blueman: blueman_2.1.2-1_amd64.deb
BlueZ: bluetoothctl: 5.53
Distribution: Xubuntu 20.04
Desktop environment: XFCE

* [x ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Hi,

I switched to Xubuntu 20.04 (from 18.04) and now I can't seem to connect correctly to my headphones (Bose QC II 35). Pairing goes fine, "Bluetooth devices" shows the headphones (and identifies it as headphones), but I only get two options on "Connect":

* Serial Port
* Don't connect

In other words, I am missing familiar "headset" that would allow me to connect. 

```
bluetoothctl     
Agent registered
[CHG] Controller 00:1A:7D:DA:71:11 Pairable: yes
[CHG] Controller 00:1A:7D:DA:71:11 Discovering: yes
[NEW] Device 4C:87:5D:CB:83:72 Moon Boots
[CHG] Device 4C:87:5D:CB:83:72 Connected: yes
[CHG] Device 4C:87:5D:CB:83:72 UUIDs: 00000000-deca-fade-deca-deafdecacaff
[CHG] Device 4C:87:5D:CB:83:72 UUIDs: 00001101-0000-1000-8000-00805f9b34fb
[CHG] Device 4C:87:5D:CB:83:72 UUIDs: 00001108-0000-1000-8000-00805f9b34fb
[CHG] Device 4C:87:5D:CB:83:72 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Device 4C:87:5D:CB:83:72 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device 4C:87:5D:CB:83:72 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device 4C:87:5D:CB:83:72 UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[CHG] Device 4C:87:5D:CB:83:72 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 4C:87:5D:CB:83:72 UUIDs: 81c2e72a-0591-443e-a1ff-05f988593351
[CHG] Device 4C:87:5D:CB:83:72 UUIDs: 931c7e8a-540f-4686-b798-e8df0a2ad9f7
[CHG] Device 4C:87:5D:CB:83:72 UUIDs: f8d1fbe4-7966-4334-8024-ff96c9330e15
[CHG] Device 4C:87:5D:CB:83:72 ServicesResolved: yes
[CHG] Device 4C:87:5D:CB:83:72 Paired: yes
[CHG] Device 4C:87:5D:CB:83:72 ServicesResolved: no
[CHG] Device 4C:87:5D:CB:83:72 Connected: no
```

Can you help me?

infirit, 2020-09-09:
There is no separate menu entry to connect an audio device anymore. You
should use plain connect entry which handles connecting several profiles
including audio/headset.

cschramm, 2020-09-09:
Are you using the setup assistant? If I'm not mistaken it lacks that generic connect entry (which makes me think about removing the assistant completely again) and you have to use the right-click or menu bar device menu in manager instead.

tracek, 2020-09-10:
Thanks for the response! 

Steps I am following:

1. Search for the device (SUCCESS).
2. Open context menu on the device, select `Setup`.
3. Pair the device (SUCCESS).
4. I get two options to connect: `Serial port` and `Don't connect`.
5. `Serial port` it fails to connect.

I am also getting [this error](https://bugs.launchpad.net/ubuntu/+source/blueman/+bug/1860851). I have similar kernel, though higher Ubuntu (specifically, Xubuntu) and Python versions.

cschramm, 2020-09-12:
As expected, in step 2 you open blueman-assistant and in step 4 it lacks the generic connect. Instead of using "setup", you can just use "pair" and then "connect" from that menu.

@infirit: What do you think? Should we get rid of assistant? Otherwise we should probably add a generic connect option to it.

tracek, 2020-09-17:
@cschramm Thanks for the patience in explaining this, somehow I have totally missed that option. Somehow I thought that "Connect" is just a label and the only option is what is below "Connect To" (which is Serial). Selecting "Connect" works, most of the time :).

I'd indeed vote for removing of the assistant. Because things didn't work there, I was assuming they don't work elsewhere.

closed 2020-09-17 by tracek