blueman-project/blueman #1381: Sep 09 18:32:34 debian bluetoothd[1081]: Loading LTKs timed out for hci0

techganesh007, 2020-09-09:
blueman:version 5.50
BlueZ:version 5.50
Distribution:Debian
Desktop environment:Wayland

 Sep 09 18:32:34 debian bluetoothd[1081]: Loading LTKs timed out for hci0

unable to start snap bluez.obex and bluez.bluez services 

some issue with d-bus configuration

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2020-09-09:
You're reporting a BlueZ issue to the blueman project. This is the wrong place and I have never seen that issue and absolutely no idea what it's about apart from what Google gives me.

closed 2020-09-09 by cschramm

techganesh007, 2020-09-12:
Dear Christopher,

Thanks for informing me.
Please share with me the correct group to raise my query.

Please more logs for your reference.

Log-
org.freedesktop.DBus.Python.FileNotFoundError: Traceback (most recent call
last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 707, in
_message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3/dist-packages/blueman/main/DbusService.py", line
36, in wrapper
    return method(*args[1:], **kwargs)
  File
"/usr/lib/python3/dist-packages/blueman/plugins/mechanism/Network.py", line
66, in ReloadNetwork
    nc.apply_settings()
  File "/usr/lib/python3/dist-packages/blueman/main/NetConf.py", line 334,
in apply_settings
    self.del_ipt_rules()
  File "/usr/lib/python3/dist-packages/blueman/main/NetConf.py", line 272,
in del_ipt_rules
    call(["/sbin/iptables", "-t", table, "-D", chain] + rule.split(" "))
  File "/usr/lib/python3.7/subprocess.py", line 323, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.7/subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.7/subprocess.py", line 1522, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/sbin/iptables':
'/sbin/iptables'

cschramm, 2020-09-12:
See http://www.bluez.org/contact/. You'll probably want to use the user mailing list.

The error is fixed in Debian package version 2.1.3-1. I assume you're using 2.0.8.