blueman-project/blueman #1385: Bluetooth manager cant detect devices - Ubuntu OS

rdprabhu2167, 2020-09-19:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I am new to Ubuntu and Github (but have n years of experience working in the Unix environment..so its all coming back to me!! My problem is ..my laptop bluetooth manager refuses to detect any bluetooth devices around. I tried on solution ..i.e copying the missing *.hcd file and copying it in the said location ..rebooting the system. Didnt work ..any suggestions ?

cschramm, 2020-09-27:
Most probably a firmware issue, yes, but could you please provide basic information and output of such a discovery process from blueman-applet so that we can take a look?

ellitrym1, 2020-10-03:
Similar issue,
It was working fine but now whenever i search for a bluetooth device, nothing shows up even tho the device is visible. Plus after hitting the search button, the button becomes unclickable unless you restart the app itself.

cschramm, 2020-10-03:
@ellitrym1: Similar issue => same information required.

Tanmay-Chavan, 2020-10-06:
the same issue, the only difference being I tried to install aptX support, and it got messed up the same as above.

cschramm, 2020-10-06:
@Tanmay-Chavan: As you demonstrated it with bluetoothctl, it's clearly not an issue with blueman.

Tanmay-Chavan, 2020-10-07:
Oh okay, thanks. I know this is tangential to the original point, but can you suggest any fix?

cschramm, 2020-10-07:
@Tanmay-Chavan: In generals it's probably related to kernel / driver / firmware. Apart from checking the syslog, I'd look up the chip you're using and commit a web search on others' experience with it and Linux (Ubuntu 20.04 I guess). From your specific description, it sounds like it used to work, then you tried to install aptX support, and then it did not work anymore. That sounds like overlooked some effect of your aptX adventures that you did not and still should revert to get it back working.

closed 2020-12-15 by github-actions[bot]