blueman-project/blueman #1388: Implementing "Reset"  in code

JulyJim, 2020-09-25:
blueman: 2.0.4 
BlueZ:  ??
Distribution: Ubuntu 16 04 
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

JulyJim, 2020-09-25:
OK , I got here 

blueman/blueman/bluez/Adapter.py 

Is that a start ?  I am not that versed in Python, but will this lead to "blueZ"  ? I prefer to eliminate "high" level wrappers  if possible.

cschramm, 2020-09-25:
I have no idea what this ticket is about. 🤷‍♂️

JulyJim, 2020-09-26:
Per your site instructions I can as for assistance, But if you do not get it , just let somebody else reply.

infirit, 2020-09-26:
> But if you do not get it , just let somebody else reply.

There are 2 of us, @cschramm and me and I also do not understand what this is about..

JulyJim, 2020-09-26:
>  OK, I need to know how  is "Remove"  implemented in " Bluetooth Manager " any references to actual code hopefully leading to "blueZ" source code.

cschramm, 2020-09-26:
It's a DBus call to `org.bluez`, interface `org.bluez.Adapter1` on the respective adapter's object path, method `RemoveDevice`.

closed 2020-09-26 by cschramm

infirit, 2020-09-26:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/adapter-api.txt

JulyJim, 2020-09-26:
BluetoothManager FYI 

Ido not know the sequence of replies I have received. 

Hereare my thoughts, in no particular order. 

Thelink referenced apparently makes no distinction between “adapter”and “device”.

However,it implies that “stopping the scan” will erase something – I amnot sure if Qt will work that way. 

Frommy research so far it looks that Qt will NOT perform physical searchAFTER the API is used first time. So my “hack” will be to NOTuse Qt for “scan for nearby devices”. 

Icannot prove it now , but I think “Bluetooth Manager” works sameway – it will enable “Scan” only once up front. Perhaps thedesigner logic was – why scan second time…things do not change (?) ... 

 Asfar as arbitrary closingmypost– it must be more important to maintaing clean slate instead ofresolving issues and letting the customer know.

Thisgives an impression you , owner of the site, do not give a hoot aboutcustomer – and do not care to receive feedback. ( So disregard whatI just posted ).
CASE CLOSED no reply necessary 

 
 

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/adapter-api.txt

cschramm, 2020-09-26:
I honestly still don't have the slightest idea what your wish is. Ticket is closed as it seemed to be to know what blueman does when you click "Remove" and we provided the simple answer.

We're not owners of this site (GitHub) and do not have customers (if you're willing to pay for our volunteer work, you're welcome to beta-test our halfway set up Open Collective https://opencollective.com/blueman).

infirit, 2020-09-26:
I think you are seriously confused what blueman is and how it interacts with the underlying kernel bluetooth subsystem.

JulyJim, 2020-09-27:
Well, thanks for the offer, but I am not interested in being insulted ( ",,,you are confused ...")  nor working with a group who cannot read  - I asked HOW  is "Reset"  implemented in "Bluetooth Manager" - simple as that .
Have a swell day.

JulyJim, 2020-09-28:
Yes, I am seriously confused.Please enlighten  me 
How does this complicated process works ? 

1. Open Bluetooth Manager.2. Push "scan" to "Scan for nearby devices" . 
3. Bluetooth Manager shows , after noticeable delays as expected - THREE discovered devices. 
4. UNPLUG ONE OF THE DISCOVERED DEVICES!5. Push "scan" AGAIN to "Scan for nearby devices" . 
6. Bluetooth Manager IMMEDIATELY , STILL  shows THREE discovered devices. 

Magic?  
Perhaps special feature  " interacting the underlying kernel bluetooth subsystem "  ? 

BUG ! 

Don't bother to fix it, you could get hurt using your voluntary  brain. 

Cheers

cschramm, 2020-09-28:
Not eager to talk to insulting, confusing people, but let's try one more time.

> 1. Open Bluetooth Manager.
> 2. Push "scan" to "Scan for nearby devices" .
> 3. Bluetooth Manager shows , after noticeable delays as expected - THREE discovered devices.
> 4. UNPLUG ONE OF THE DISCOVERED DEVICES!

Unplug WHAT from WHERE? (Sorry for the caps, I'm just trying to adapt a little bit.)

> 5. Push "scan" AGAIN to "Scan for nearby devices" .
> 6. Bluetooth Manager IMMEDIATELY , STILL  shows THREE discovered devices.

blueman-manager ALWAYS shows ALL devices currently provided by BlueZ. Unless you REMOVED one, AWAITED THE TIMEOUT after which BlueZ removes temporary devices automatically (should be three minutes, see https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/adapter.c?id=36441825004c70c81ff72915c51e0e60b30d91ff#n77), or DISCOVERED A NEW ONE during the second scan, it is more than expected behavior of BlueZ that nothing changed. If you want to tell people over there, that that's a bug, please see http://www.bluez.org/contact/, but consider that you might just insult and confuse them.

JulyJim, 2020-09-29:
How irrelevant can you get ?CAN YOU READ ?Please STOP replying with nonsense.Have a great day .