blueman-project/blueman #1398: Feature request: Connect to device while scanning

Thalley, 2020-10-05:
In `blueman-manager` GUI version 2.1.3: If you are  explicitly scanning for devices, and you find a connectable device, then attempting to connect to that device will fail since the scan is ongoing. 

Attempting to connect to a device via the GUI while scanning should instead stop the scanning, and attempt a connect instead.

infirit, 2020-10-05:
I would agree but please can you post your blueman-applet log that shows scanning and the subsequent connect attempt.

Thalley, 2020-10-05:
> I would agree but please can you post your blueman-applet log that shows scanning and the subsequent connect attempt.

@infirit Sure, can you tell me how/where to find that log? `blueman-report` returns a 401.

infirit, 2020-10-07:
See https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-blueman

Thalley, 2020-10-07:
Ran as `blueman-applet --loglevel debug`:

```
blueman-applet 17.57.40 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci1 {'Discovering': True}
blueman-applet 17.57.40 DEBUG    DiscvManager:82 on_adapter_property_changed: prop Discovering True
blueman-applet 17.57.42 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci1/dev_4C_87_5D_2A_59_EC {'Address': '4C:87:5D:2A:59:EC', 'AddressType': 'public', 'Name': 'Bose NC 700 Headphones', 'Alias': 'Bose NC 700 Headphones', 'Class': 2360344, 'Icon': 'audio-card', 'Paired': True, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['00000000-deca-fade-deca-deafdecacaff', '00001101-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '0000110d-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '0000111e-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '81c2e72a-0591-443e-a1ff-05f988593351', '85dbf2f9-73e3-43f5-a129-971b91c72f1e', '931c7e8a-540f-4686-b798-e8df0a2ad9f7', 'f8d1fbe4-7966-4334-8024-ff96c9330e15'], 'Modalias': 'bluetooth:v009Ep4024d0170', 'Adapter': '/org/bluez/hci1', 'ManufacturerData': {3843: [209, 18, 170, 122, 128, 180, 157, 119, 26, 114, 73]}, 'ServiceData': {'0000fe2c-0000-1000-8000-00805f9b34fb': [205, 130, 86]}, 'ServicesResolved': False}
blueman-applet 17.57.42 INFO     RecentConns:161 notify    : <Device.Device object at 0x7f72a63c3d80 (blueman+bluez+Device+Device at 0x560c8cfc8760)> 00000000-0000-0000-0000-000000000000
blueman-applet 17.57.42 INFO     RecentConns:91 initialize: rebuilding menu
blueman-applet 17.57.50 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci1 {'Discovering': False}
blueman-applet 17.57.50 DEBUG    DiscvManager:82 on_adapter_property_changed: prop Discovering False

```

The log start from when I press "Search", and while searching I attempt to connect to device (`4C:87:5D:2A:59:EC`), and then the log ends with the scan timeout.

cschramm, 2020-11-02:
Connecting works in the first case and fails in the second, right?

What I see in the logs is that the discovered device gets removed and re-added right after you try to connect to it. That's completely unexpected and I cannot reproduce it on my side (with an LE mouse). What I'm missing, though, is a `do_g_properties_changed: /org/bluez/hci1/dev_4C_87_5D_2A_59_EC` after the device gets created.

Also, hci enumeration typically starts at 0, do you have an hci0 as well, i. e. multiple adapters?

We might find out a reason why the device gets re-added by checking the bluetoothd debugging output of the same sequence. The easiest way is to stop the `bluetooth.service` and run `bluetoothd -d -n` (as root and with the correct path) in a terminal.

Which BlueZ version are you using? You skipped our standard information template.

cschramm, 2020-11-02:
Oh, and: You're trying to connect to the new device without pairing. Is that supported by the device? The exact same sequence works fine if you first await the the discovery to finish, right?

Thalley, 2020-11-02:
Hi @cschramm , must've missed the template when creating the bug. 

To answer your questions:
Yes, I do have multiple adapters (working on the Bluetooth stack of Zephyr)

The device (a BT headset) does support pairing, and is always paired when I use it

Neither the first case or the second cast connects successfully. 

`bluetoothctl --version` outputs `bluetoothctl: 5.55`, but this issue have actually existed for me for many months. 

I'll try to get around an providing some logs from `bluetoothd -d -n`.

cschramm, 2020-11-03:
The question was if the device allows connecting *without* pairing as that seems to be what you're trying in the second log.

So in the first log you're trying to connect to the known and paired device during an active discovery and just nothing happens, right? If you wait for the discovery to stop, connecting works fine? I'll play with my devices a little bit but I don't think I can reproduce that at all. It might be a limitation of your adapter / firmware as neither blueman nor BlueZ should do anything that would prevent connecting to work during a discovery.

The multiple adapters setting concerns me a bit but I currently don't see how it could be related.

Thalley, 2020-11-03:
@cschramm I just tried connecting to a paired device while scanning with btmon running. Evidently the create connection command is sent, without stopping LE scanning. I cannot find anything explicitly stating that this is disallowed in the Bluetooth Core Spec, so I'll try to reach out to Intel (my controller is an Intel controller) to verify this part. 

```
 MGMT Command: Start Discovery (0x0023) plen 1                                  {0x0001} 11.276519
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
< HCI Command: LE Set Random Address (0x08|0x0005) plen 6                             #52 11.276677
        Address: 2B:D2:F4:F7:7D:B6 (Non-Resolvable)
> HCI Event: Command Complete (0x0e) plen 4                                           #53 11.386691
      LE Set Random Address (0x08|0x0005) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                            #54 11.386784
        Type: Active (0x01)
        Interval: 22.500 msec (0x0024)
        Window: 11.250 msec (0x0012)
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                           #55 11.387688
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                #56 11.387770
        Scanning: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                           #57 11.389690
      LE Set Scan Enable (0x08|0x000c) ncmd 2
        Status: Success (0x00)
< HCI Command: Inquiry (0x01|0x0001) plen 5                                           #58 11.389775
        Access code: 0x9e8b33 (General Inquiry)
        Length: 10.24s (0x08)
        Num responses: 0
> HCI Event: Command Status (0x0f) plen 4                                             #59 11.390684
      Inquiry (0x01|0x0001) ncmd 2
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                                   {0x0001} 11.390784
      Start Discovery (0x0023) plen 1
        Status: Success (0x00)
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
@ MGMT Event: Discovering (0x0013) plen 2                                        {0x0001} 11.390810
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
        Discovery: Enabled (0x01)
> HCI Event: LE Meta Event (0x3e) plen 43                                             #60 12.354705
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Public (0x00)
        Address: 7C:64:56:A1:BC:C1 (Samsung Electronics Co.,Ltd)
        Data length: 31
        Flags: 0x1a
          LE General Discoverable Mode
          Simultaneous LE and BR/EDR (Controller)
          Simultaneous LE and BR/EDR (Host)
        Company: Samsung Electronics Co. Ltd. (117)
          Data: 420401016f7c6456a1bcc17e6456a1bcc00d000000000000
        RSSI: -84 dBm (0xac)
> HCI Event: LE Meta Event (0x3e) plen 40                                             #61 12.355657
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Scan response - SCAN_RSP (0x04)
        Address type: Public (0x00)
        Address: 7C:64:56:A1:BC:C1 (Samsung Electronics Co.,Ltd)
        Data length: 28
        Name (short): [TV] Samsung 6 Series (55)
[... 85 lines cut ...]
        Address: 4C:87:5D:2A:59:EC (Bose Corporation)
        Packet type: 0xcc18
          DM1 may be used
          DH1 may be used
          DM3 may be used
          DH3 may be used
          DM5 may be used
          DH5 may be used
        Page scan repetition mode: R2 (0x02)
        Page scan mode: Mandatory (0x00)
        Clock offset: 0x0000
        Role switch: Allow slave (0x01)
> HCI Event: Command Status (0x0f) plen 4                                             #69 14.642602
      Create Connection (0x01|0x0005) ncmd 1
        Status: Success (0x00)
> HCI Event: Connect Complete (0x03) plen 11                                          #70 14.643649
        Status: Command Disallowed (0x0c)
        Handle: 65535
        Address: 4C:87:5D:2A:59:EC (Bose Corporation)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
@ MGMT Event: Connect Failed (0x000d) plen 8                                     {0x0001} 14.643726
        BR/EDR Address: 4C:87:5D:2A:59:EC (Bose Corporation)
        Status: Busy (0x0a)
```

cschramm, 2020-11-03:
I actually *am* able to reproduce this using a paired phone and paired speakers.

```
> HCI Event: Command Status (0x0f) plen 4                                                         #465 [hci0] 96.801858
      Create Connection (0x01|0x0005) ncmd 1
        Status: Success (0x00)
> HCI Event: Connect Complete (0x03) plen 11                                                      #466 [hci0] 96.802816
        Status: Command Disallowed (0x0c)
        Handle: 65535
        Address: CC:9F:7A:A5:47:34 (Chiun Mai Communication Systems, Inc)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
@ MGMT Event: Connect Failed (0x000d) plen 8                                                  {0x0001} [hci0] 96.802901
        BR/EDR Address: CC:9F:7A:A5:47:34 (Chiun Mai Communication Systems, Inc)
        Status: Busy (0x0a)
= bluetoothd: profiles/audio/avdtp.c:avdtp_connect_cb() connect to CC:9F:7A:A5:47:34: Device or resource...   96.803533
```

Adapter is Intel as well, 8087:0029 AX200.

Thalley, 2020-11-03:
> I actually _am_ able to reproduce this using a paired phone and paired speakers.
>
> ```
>       Create Connection (0x01|0x0005) ncmd 1
>         Status: Success (0x00)
>         Status: Command Disallowed [...]

Interesting, I have the same AX200 controller. This might indicate a problem on Intel's side then. 

I'll try to see what the local supported features are, because there should be a bit stating whether or not it can perform simultaneous scan and connection. It might either report support for that, while not being able to do that, or perhaps Bluez isn't reading that bit correctly.

Thalley, 2020-11-03:
The controller reports support for simultaneous initiating and scanning state, and should not report command disallowed when this happens. I guess we can close this issue and then I'll forward the bug to Intel, unless @cschramm have anything more to add?

Thalley, 2020-11-03:
Btw, figured it was an issue in the host, since it works fine in Windows, but I suppose Windows just always stops scan first or something.

cschramm, 2020-11-03:
Same thing with 8087:07dc. The Connect Complete HCI event with status Command Disallowed is either coming from the firmware or the controller itself. Either the Windows firmware differs or, as you said, Windows might stop the discovery when trying to connect.

We could actually think about doing that as well, if this is a generic issue, but I'd like to have this verified with other (non-Intel) controllers first. I got a cheap 0c10:0000 controller but unfortunately it seems like I'd have to hack my kernel to make it work :see_no_evil: (current Linux treats it as Silcone Wave and it looks like that but it does not work properly and could theoretically be Canyon or Broadcom 203x or whatever as well...).

Thalley, 2020-11-03:
@cschramm I have a couple of different controllers I can try with. I know one of them doesn't support simultaneous scan and connect when reporting the supported states, so that would also verify whether or not `blueman` handles that correctly. Will report back with my findings.

Thalley, 2021-01-18:
@cschramm Finally got in contact with Intel. For the AX200 there should be a firmware update coming at some point to fix that. For the controller I use, Intel 8260, it's too old to get any new firmware updates it seems. 

We could perhaps do what we assume Windows is doing, and simply cancel scan before attempting to create any connection.

cschramm, 2021-01-20:
Hm. Sounds really edgy to me, but I guess stopping the discovery shouldn't do much harm. What did they say? Is it by chance that exactly those two chips behave that way or is it actually kind of all Intel wireless adapters out there? :facepalm:

Thalley, 2021-01-20:
> Hm. Sounds really edgy to me, but I guess stopping the discovery shouldn't do much harm. What did they say? Is it by chance that exactly those two chips behave that way or is it actually kind of all Intel Bluetooth adapters out there?

Since there's roughly 5 years between the two controller (Intel 8260 and AX200), I think it's by chance. The firmware they had for the AX200 wouldn't work on the 8260 I was told. 

I don't think it's the same issue for all Intel controllers, as I don't have this issue on my laptop which also uses an Intel controller (don't have the name of that controller right now).

cschramm, 2021-01-20:
Interesting that I hit exactly those two (I suppose my 8087:07dc is that 8260). With other controllers, the discovery just continues while pairing / connecting takes place in parallel? I see basically three options to stop the discovery before connecting:

- Just always.
- For specific controllers. I don't think we get anything more useful than the hardware address, though, so that's not really feasible.
- If we get EAGAIN on first try, then retry after stopping. We already implemented a way to catch that with https://github.com/blueman-project/blueman/commit/ed8d903f8a5dfc9a6966bae78d16b0d1998f120e.

Thalley, 2021-01-20:
> * Just always.

That's the easy and foolproof solution. What (negative) consequences could that have? 

> * For specific controllers. I don't think we get anything more useful than the hardware address, though, so that's not really feasible.

It's a possibility, although I'm not sure how many people would add any potential problematic controllers to such a list. I guess most people would just try with their controller, complain it doesn't work, and perhaps try another BT app. 

> * If we get EAGAIN on first try, then retry after stopping. We already implemented a way to catch that with [ed8d903](https://github.com/blueman-project/blueman/commit/ed8d903f8a5dfc9a6966bae78d16b0d1998f120e).

Also a good approach. If I have a bad controller, I don't care much whether the first try fails, as long as it eventually succeeds (within a certain timeframe of course...)

cschramm, 2021-01-21:
Untested draft in #1485. :slightly_smiling_face:

closed 2021-04-17 by cschramm

cschramm, 2022-08-02:
@Thalley: Is this still an issue today? Just wondering because I think our handling would not work with BlueZ 5.62+.

Thalley, 2022-08-08:
> @Thalley: Is this still an issue today? Just wondering because I think our handling would not work with BlueZ 5.62+.

Just tried, and at least connecting to an already paired device works while scanning.

Thalley, 2022-08-09:
> @Thalley: Is this still an issue today? Just wondering because I think our handling would not work with BlueZ 5.62+.

Tested with BlueZ 5.64 today and it seems to work (i.e. I can connect while scanning).

cschramm, 2022-08-15:
I do not seem to be able to replicate this at all. I expected it to show up with a fresh Debian 10 installation and the `8087:07dc` chip but at least with USB passthrough it did not. :confused: Maybe a point release of the 4.9 kernel in a point release of Debian 10 makes the difference there. I also was not able to replicate it with Manjaro downgrading Linux, firmware and BlueZ to the lowest available versions (I got some EAGAIN but more randomly than depending on the discovery state of the adapter; that was with USB pass-through as well).

Anyway, it seems like this is not a generic issue with recent Linux and firmware anymore and the error is not passed by recent BlueZ so I guess we can just drop our mitigation.

PR #1485 (2021-04-17): Stop discovery and retry connection on initial EAGAIN
There are drivers that do not support connecting while a discovery is still active.

Closes #1398
commit 358dbddb (2021-04-17): Stop discovery and retry connection on initial EAGAIN

There are drivers that do not support connecting while a discovery is still active.

Closes #1398