blueman-project/blueman #1400: Feature request: Disable one bluetooth device

samuk190, 2020-10-10:
Thats it

samuk190, 2020-10-12:
sorry if i didnt asked politely ;) I love the work you guys do!

infirit, 2020-11-06:
Foolish user!! Being rude and demanding shit 😜 .

I don't see a simple way to block one adapter. The only option I see right now is to selectively (un)power adapters.

A workaround might be to use the kernel's bind/unbind feature, see https://unix.stackexchange.com/questions/279283/how-to-prevent-a-kernel-module-from-opening-a-specific-device

samuk190, 2020-11-06:
> Foolish user!! Being rude and demanding shit stuck_out_tongue_winking_eye .
> 
> I don't see a simple way to block one adapter. The only option I see right now is to selectively (un)power adapters.
> 
> A workaround might be to use the kernel's bind/unbind feature, see https://unix.stackexchange.com/questions/279283/how-to-prevent-a-kernel-module-from-opening-a-specific-device

hehe >.< 
That's a good idea, I guess its the same approach of the bluetooth manager of KDE.

cschramm, 2022-09-08:
Trying to wrap my head around this with the current status quo.

I think there are three levels of disabling a Bluetooth adapter: Kernel driver, rfkill and BlueZ power state.

Preventing the kernel from connecting the device to the driver is done with the `unbind` interface. I don't think that's in blueman's scope.

I'm not sure if rfkill actually works. Do multiple adapters always have dedicated killswitches that you could set? :thinking:

Regarding the power state, I consider BlueZ' `AutoEnable` which is enabled by default in 5.65 the new normal. It's thus pointless to unpower a specific adapter as bluetoothd will just re-power it.