blueman-project/blueman #1402: Context menu key doesn't work

niekomaatjes, 2020-10-14:
blueman: 2.1.3
BlueZ: 
Distribution: Debian GNU/Linux rodete
Desktop environment: Cinnamon

* [ ✓ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

To connect to a Bluetooth device, I can right-click and choose Connect. This works fine. Ideally, I'd like to be able to do this without using my mouse, but e.g. using the context menu key on my keyboard. Unfortunately, this doesn't do anything.

Could you please add support for the context menu key, and/or add other keyboard shortcuts to connect/disconnect?

infirit, 2020-10-14:
There is a missing shortcut but but you can workaround it.

The device menu in the menubar is the exact same menu when you right click. The following is language specific so the actual key may be different but holding the alt key will reveal it. In English hold the alt and press d, then hit enter as the connect menu entry is already highlighted.

#1404 adds the c as shortcut key.

niekomaatjes, 2020-10-14:
Thank you for the quick resolution! The menu will indeed do for now :)

cschramm, 2020-10-14:
@niekomaatjes: Could you try #1405? :slightly_smiling_face:

niekomaatjes, 2020-10-15:
Unfortunately not :/ On Debian rodete I get "No module named 'blueman'", and haven't yet figured out how to get around that. https://github.com/blueman-project/blueman/issues/875 discusses that a bit, but that didn't get me very far.

cschramm, 2020-10-15:
It's already merged and in the snapshot on https://blueman.cschramm.eu/debian/ now. Assuming (I don't know much about it :shrug:) rodete uses Debian testing (and thus has Python 3.8) and your architecture is amd64, [this](https://blueman.cschramm.eu/debian/pool/main/b/blueman/blueman_2.2-git1602781736r845382d0-python3.8-1_amd64.deb) is the direct link to the deb as long as nothing happens on master.