blueman-project/blueman #1412: Pairing process not prompting PIN

enriquemorenotent, 2020-10-29:
blueman: 2.1.2
BlueZ:
Distribution: Ubuntu 20.04
Desktop environment: Gnome

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I have been trying to connect to my Logitech MX Keys keyboard without success. For some reason, the prompt to write the PIN is not comming up. I thought I would go a little more low-level and I tried the following:

```
➜ bluetoothctl connect CB:DD:B4:5C:4E:E3
Attempting to connect to CB:DD:B4:5C:4E:E3
[CHG] Device CB:DD:B4:5C:4E:E3 Connected: yes
Connection successful

~ took 8s 
➜ bluetoothctl trust CB:DD:B4:5C:4E:E3
[CHG] Device CB:DD:B4:5C:4E:E3 Trusted: yes
Changing CB:DD:B4:5C:4E:E3 trust succeeded

~ 
➜ bluetoothctl pair CB:DD:B4:5C:4E:E3
Attempting to pair with CB:DD:B4:5C:4E:E3
[CHG] Device CB:DD:B4:5C:4E:E3 ServicesResolved: yes
[CHG] Device CB:DD:B4:5C:4E:E3 ServicesResolved: no
[CHG] Device CB:DD:B4:5C:4E:E3 Connected: no
Failed to pair: org.bluez.Error.AuthenticationCanceled

~ took 30s 
```

Even doing this did not prompt me to input a PIN number. Why could this be?

cschramm, 2020-10-30:
Assuming that blueman-applet is running and gnome-bluetooth is not, output from blueman-applet should help identify what happens when you try to pair.

You can also try what happens when using a bluetoothctl agent (commands `agent on` and `default-agent`, then try to pair).

enriquemorenotent, 2020-10-30:
There is no output in blueman-applet. Unless you mean some sort of logfile...

It didn't work with an agent neither

cschramm, 2020-10-30:
You need to run `blueman-applet --loglevel debug` from a terminal (after exiting any running instance) as described on the troubleshooting page.

enriquemorenotent, 2020-10-30:
The logs do not update at all, after starting the pairing process

cschramm, 2020-10-30:
That does not sound like a blueman issue then. Please check the BlueZ agent logs. `sudo systemctl stop bluetooth.service` to stop bluetoothd and then `sudo bluetoothd -d src/agent.c -n` to run it in foreground with agent debug logs should do the trick. You should usually see a `src/agent.c:agent_request_confirmation()` line when pairing that gives information on the receiving agent.

enriquemorenotent, 2020-10-30:
I have run it in the foreground, and this is all I see when I start the pairing process:

```
bluetoothd[187555]: src/agent.c:agent_ref() 0x556d5dd53ca0: ref=2
bluetoothd[187555]: src/agent.c:agent_ref() 0x556d5dd53ca0: ref=3
bluetoothd[187555]: src/agent.c:agent_unref() 0x556d5dd53ca0: ref=2
bluetoothd[187555]: No cache for CB:DD:B4:5C:4E:E3
bluetoothd[187555]: src/agent.c:agent_unref() 0x556d5dd53ca0: ref=1
```

Nothing more

cschramm, 2020-11-01:
bluetoothd does not trigger an agent at all then. Probably it does not even receive any command to do so. There might be more from `src/adapter.c` (or just `-n` to get output from all files).

enriquemorenotent, 2020-11-01:
What I posted 2 days ago was already using the `-n` flag

cschramm, 2020-11-01:
:thinking: So it looks like there's absolutely no prompt for anything.

Different approach: All paths that lead to org.bluez.Error.AuthenticationCanceled in bluetoothd include some debugging output at least from src/device.c that can be used to isolate what's going on, but I guess ultimately it will be kernel / firmware / remote device behavior.

enriquemorenotent, 2020-11-01:
So, what should I do? You have added the tag "question", but I am not sure what is it that you are asking now.

cschramm, 2020-11-01:
Don't care about the label, it's just meant to signal that you're asking a question that you need help with (unless some important piece is still missing that turns this into a bug).

You should check the full debugging output of bluetoothd when trying to pair to isolate the problem. If you need help with interpreting it, you're welcome to provide it here.  (If you have the possibility I would try a different kernel / firmware / adapter.)

I found https://bbs.archlinux.org/viewtopic.php?id=258125 where two guys seem to successfully pair the same keyboard using a bluetoothctl agent in Arch Linux with probably a very similar kernel to yours. The initial post sounds like it did not work with blueman either, so there might actually be some issue but the fact that we're registering an agent with full capabilities and that your bluetoothd output does not even shows a try to ask the agent to do something seems like blueman does not even get a chance to do something wrong.

enriquemorenotent, 2020-11-01:
Well, yeah. I have no idea how to proceed now, so if anyone can provide help, I would be very thankful

cschramm, 2020-11-02:
Let me rephrase all directions that I gave you:

* Stop bluetoothd and start it manually with `sudo bluetoothd -d -n`, then try to pair and analyze or provide all debug output after you started pairing to identify the specific cause of the org.bluez.Error.AuthenticationCanceled. It probably also makes sense to check for kernel messages (dmesg).
* Open bluetoothctl, use `agent on`, `default-agent`, `scan on`, wait until the device got discovered, use `pair CB:DD:B4:5C:4E:E3` and see if a passkey shows up in bluetoothctl just as described in the Arch Linux forums.
* If you know how to use a different kernel or firmware version, try an upgrade / downgrade to see if the problem sticks.
* If you have another Bluetooth adapter (USB) at hand, try the whole thing with that one to see if you encounter the same problem.

enriquemorenotent, 2020-11-02:
I found the solution. Some firmware was missing:

https://askubuntu.com/questions/1270419/20-04-logitech-mx-keys-wont-pair-with-bluetooth

Thanks

closed 2020-11-02 by cschramm