blueman-project/blueman #1423: blueman will only start as root

juanitotc, 2020-11-04:
blueman: 2.1.4
BlueZ: 5.54
Distribution: piCore64 (aarch64)
Desktop environment: flwm

blueman fails to start (Could not connect: Connection refused (39)) without

```
$ sudo blueman-applet
$ sudo blueman-manager
```

This is a non-systemd distro, what needs to be changed so blueman runs with user privileges?

cschramm, 2020-11-04:
Nothing fundamental changed in 2.1.4. Sounds like you have your #206 again. :wink: Maybe #1086 could be helpful.

juanitotc, 2020-11-05:
I'd completely forgotten about #206, but it's not that - the shebangs in /usr/local/bin and libexec all correctly reference /usr/local/bin/python3.

#1806 is a systemd distro, so I don't think it is relevant.

cschramm, 2020-11-05:
Then I'll need more context to be able to assist, I guess. :shrug:

juanitotc, 2020-11-05:
Sorry - I thought I'd posted something already.

```
$ blueman-applet
blueman-applet version 2.1.4 starting
Traceback (most recent call last):
  File "/usr/local/bin/blueman-applet", line 42, in <module>
    BluemanApplet()
  File "/usr/local/lib/python3.8/site-packages/blueman/main/Applet.py", line 30, in __init__
    self.DbusSvc = DbusService("org.blueman.Applet", "org.blueman.Applet", "/org/blueman/applet",
  File "/usr/local/lib/python3.8/site-packages/blueman/main/DbusService.py", line 27, in __init__
    self._bus = Gio.bus_get_sync(bus_type)
gi.repository.GLib.Error: g-io-error-quark: Could not connect: Connection refused (39)
```

cschramm, 2020-11-05:
Looks like your user does not have a proper D-Bus session bus running that he could connect to. That is completely independent from anything blueman specific (it's just the generic `Gio.bus_get_sync(Gio.BusType.SESSION)` that fails).

juanitotc, 2020-11-06:
Ah - either dbus was not started before the GUI or it stopped and restarted once in the GUI.

If I ensure dbus is started before the GUI, then blueman works as expected.

Sorry for the noise.

closed 2020-11-06 by juanitotc