blueman-project/blueman #1430: obexd cannot be started under Debian with latest bluez-obexd (5.55-1)

akoun, 2020-11-06:
blueman: 2.3.1
BlueZ: 5.55-1
Distribution: Debian GNU/Linux bullseye/sid
Desktop environment: XFCE

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2020-11-06:
I cannot confirm that.

`systemctl status --user dbus-org.bluez.obex.service`?

akoun, 2020-11-08:
> systemctl status --user dbus-org.bluez.obex.service
> Unit dbus-org.bluez.obex.service could not be found.

I think the following is relevant as well
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=972070

cschramm, 2020-11-08:
Sounds like an issue with that Debian package then. Definitely not blueman specific as the reporter uses bluedevil. Also, I'm using Debian and bluez-obexd 5.55-1 works for me, so it is not a generic issue.

closed 2020-11-09 by cschramm